projectacrn/acrn-hypervisor #1986: UOS will hang after triggering watchdog reset

mwang106, 2018-12-05:
1.Environment
--------------------------------------------------------------
Installed version: 26550

2.Reproduce step:
--------------------------------------------------------------
1). Boot to SOS
2). SSH to SOS.
3). modify launch_UOS.sh add 
    -s 8,wdt-i6300esb \
4). ./launch_UOS.sh to boot uos.
5).Set watchdog timeout  3s "wdctl -s 3"
6). open /dev/watchdog for more than 3 seconds by using python shell as below 
>>> open('/dev/watchdog')
<_io.TextIOWrapper name='/dev/watchdog' mode='r' encoding='UTF-8'>

3. Current result:
--------------------------------------------------------------
uos will crash after triggle watchdogd reset.

4. Expected result:
--------------------------------------------------------------
after 3 seconds, system have to start rebooting.

5. repetition rate
--------------------------------------------------------------
100%

6.Logs:
--------------------------------------------------------------
root@clr-514f2d59482a401991ba01fff774fd9d ~ # wdctl -s 3
Timeout has been set to 3 seconds.
Device:        /dev/watchdog
Identity:      i6300ESB timer [version 0]
Timeout:        3 seconds
Pre-timeout:    0 seconds
FLAG           DESCRIPTION               STATUS BOOT-STATUS
KEEPALIVEPING  Keep alive ping reply          1           0
MAGICCLOSE     Supports magic close char      0           0
SETTIMEOUT     Set timeout (in seconds)       0           0
root@clr-514f2d59482a401991ba01fff774fd9d ~ #
root@clr-514f2d59482a401991ba01fff774fd9d ~ #
root@clr-514f2d59482a401991ba01fff774fd9d ~ #
root@clr-514f2d59482a401991ba01fff774fd9d ~ #
root@clr-514f2d59482a401991ba01fff774fd9d ~ # python wd.py
[   99.159921] watchdog: watchdog0: watchdog did not stop!
root@clr-514f2d59482a401991ba01fff774fd9d ~ #

mwang106, 2018-12-05:
[External_System_ID] ACRN-2006

Mingyuan18, 2018-12-24:
Can't reproduce this issue in below build:
Hypervisor: 2018w50.3.140000p_131
SOS:26800

closed 2018-12-24 by Mingyuan18