projectacrn/acrn-hypervisor #1987: UOS will have same mac address after launching UOS with virio-net

mwang106, 2018-12-05:
Environment:
HW/Board：APL UP2

Version: 26550

steps:
1:boot sos
2:APL NUC and APL UP2 boot uos with the default script
root@robin-nuc /home # ./launch_uos.sh -V 1
cpu1 online=1
cpu2 online=1
cpu3 online=1
passed gvt-g optargs low_gm 64, high_gm 448, fence 8

3:UOS has a same MAC address in APL NUC and APL UP2, So have the same IP address.
APL NUC's Information:
root@clr-da2b467367eb45fdbbbdca81b1664b39 ~ # ifconfig
enp0s4 Link encap:Ethernet HWaddr 00:16:3E:51:83:3E
inet addr:10.240.179.39 Bcast:10.240.179.127 Mask:255.255.255.128
inet6 addr: fe80::216:3eff:fe51:833e/64 Scope:Link
UP BROADCAST RUNNING MULTICAST MTU:1500 Metric:1
RX packets:87 errors:0 dropped:0 overruns:0 frame:0
TX packets:64 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000
RX bytes:75622 (73.8 Kb) TX bytes:5238 (5.1 Kb)
APL UP2's Information:
root@clr-da2b467367eb45fdbbbdca81b1664b39 ~ # ifconfig
enp0s4 Link encap:Ethernet HWaddr 00:16:3E:51:83:3E
inet addr:10.240.179.39 Bcast:10.240.179.127 Mask:255.255.255.128
inet6 addr: fe80::216:3eff:fe51:833e/64 Scope:Link
UP BROADCAST RUNNING MULTICAST MTU:1500 Metric:1
RX packets:245 errors:0 dropped:0 overruns:0 frame:0
TX packets:69 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000
RX bytes:94888 (92.6 Kb) TX bytes:5906 (5.7 Kb)

Expected result:
Step 3, UOS has a different MAC address in APL NUC and APL UP2

Actual result:
Step 3, UOS has a same MAC address in APL NUC and APL UP2 

Reproduce rate: 100%

Debugging info:
please see the attachment.

mwang106, 2018-12-05:
[External_System_ID] ACRN-1989

Mingyuan18, 2018-12-26:
Can't reproduce this issue in below build:
Clearlinux Version: 26850
 Hypervisor: 2018w51.2.140000p

closed 2018-12-26 by Mingyuan18