projectacrn/acrn-hypervisor #1991: Type “Enter” is useless in UART Console for corner case

mwang106, 2018-12-05:
1.Environment
--------------------------------------------------------------
Installed version: 26550
Version URL: http://linux-ftp.sh.intel.com/pub/mirrors/clearlinux/update
Content URL: http://linux-ftp.sh.intel.com/pub/mirrors/clearlinux/update

2.Reproduce step:
--------------------------------------------------------------
1). Boot to SOS
2). SSH to SOS.
3). ./launch_UOS.sh to boot uos.
4). minicom -s dev/ttyUSB0 in host.
5). sos_console 0 to launch sos.

3. Current result:
--------------------------------------------------------------
in step 5: It could't input(only output) in console

4. Expected result:
--------------------------------------------------------------
in step 5: It is normally in console.

5. repetition rate
--------------------------------------------------------------
100%

6.Logs:

mwang106, 2018-12-05:
[External_System_ID] ACRN-2011

Mingyuan18, 2019-01-21:
This issue of third party, so close it.

closed 2019-01-21 by Mingyuan18