projectacrn/acrn-hypervisor #1996: [APL/KBL/UP2]There is an error log when using "acrnd&" to boot uos.

mwang106, 2018-12-06:
1.Environment
--------------------------------------------------------------
Installed version: 26550
Version URL: http://linux-ftp.sh.intel.com/pub/mirrors/clearlinux/update
Content URL: http://linux-ftp.sh.intel.com/pub/mirrors/clearlinux/update

2.Reproduce step:
--------------------------------------------------------------
1). Boot to SOS
2). SSH to SOS
3). Add a Laag UOS into list over SOS SSH by
 acrnctl add [uos_bash_script]
   ex: acrnctl add launch_UOS.sh -V 1
4). Start that VM by below command
    acrnd &

3. Current result:
--------------------------------------------------------------
There is an error:Failed to open the socket(sos-lcs) to query the reason for the wake-upActivating all vms when acrnd & to boot uos.
and actually uos can boot up successfully.
4. Expected result:
--------------------------------------------------------------
Uos could boot up successfully without any error.

5. repetition rate
--------------------------------------------------------------
100%

6.Logs:
--------------------------------------------------------------
root@sos-dom0 ~ # acrnctl list
vm1 stopped
root@sos-dom0 ~ # acrnd &
[1] 638
root@sos-dom0 ~ # connect_to_server 434
Failed to open the socket(sos-lcs) to query the reason for the wake-upActivating all vms
polling 3...
Listening 3...

mwang106, 2018-12-06:
[External_System_ID] ACRN-2010

Mingyuan18, 2019-03-25:
The bug cannot reproduced with : 
Installed version: 28340
Daily tag:”2019w11.5.140000p”

closed 2019-03-25 by Mingyuan18