projectacrn/acrn-hypervisor #1999: [APLNUC][KBLNUC][APLUP2]UOS reset fails with acrnctl reset command

mwang106, 2018-12-06:
1.Environment
--------------------------------------------------------------
Installed version: 26550
Version URL: http://linux-ftp.sh.intel.com/pub/mirrors/clearlinux/update
Content URL: http://linux-ftp.sh.intel.com/pub/mirrors/clearlinux/update

2.Reproduce step:
--------------------------------------------------------------
1). Boot to SOS
2). SSH to SOS
3). Add a Laag UOS into list over SOS SSH by
 acrnctl add [uos_bash_script]
   ex: acrnctl add launch_UOS.sh -V 1
4). Start that VM by below command
    acrnctl start [vm name]
  Notes: use "acrnctl list" to check the vm name, if you don't know it. 
    ex: acrnctl start vm1-13:ce:dd
5). Over console reset vm by below command
    acrnctl reset [vm name]
    Notes: use "acrnctl list" to check the vm name, if you don't know it. 

3. Current result:
--------------------------------------------------------------
Uos cannot boot up successfully.

4. Expected result:
--------------------------------------------------------------
Uos could boot up successfully.

5. repetition rate
--------------------------------------------------------------
100%

6.Logs:
--------------------------------------------------------------
root@sos-dom0 ~ # acrnctl reset vm1
Error happens when try to stop vm. errno(-1)
Failed to stop vm1 in 10 sec

mwang106, 2018-12-06:
[External_System_ID] ACRN-2002

Mingyuan18, 2019-03-07:
The bug cannot reproduced with 
swupd info
Installed version: 28100
Version URL: https://cdn.download.clearlinux.org/update/
Content URL: https://cdn.download.clearlinux.org/update/

closed 2019-03-07 by Mingyuan18