projectacrn/acrn-hypervisor #2000: [APLNUC][IO][LaaG] After launch UOS with Audio pass-though, Device (I2C0) not exists in UOS DSDT.dsl

mwang106, 2018-12-06:
Environment:
HW/Board：APL NUC

Build link:
https://download.clearlinux.org/image/clear-26550-kvm.img.xz

Image info:
IOC version: NUC has not support IOC.
SOS version:26550
SOS kernel version:Linux sos-dom0 4.19.0-26.iot-lts2018-sos #1 SMP PREEMPT Wed Nov 28 01:10:17 UTC 2018 x86_64 GNU/Linux
UOS Type: LaaG
UOS version: 26550
UOS kernel version:Linux clr-da2b467367eb45fdbbbdca81b1664b39 4.19.0-26.iot-lts2018 #1 SMP Wed Nov 28 01:08:50 UTC 2018 x86_64 GNU/Linux

steps:
1:boot sos and enable VT-d

2:hide audio in sos
1): find out audio: $lspci 
2):Find out the numeric PCI ID for audio:$lspci -n
3):edit parameters：
#for audio device
echo "8086 5a98" > /sys/bus/pci/drivers/pci-stub/new_id
echo "0000:00:0e.0" > /sys/bus/pci/devices/0000:00:0e.0/driver/unbind
echo "0000:00:0e.0" > /sys/bus/pci/drivers/pci-stub/bind
-s n,passthru,0/0e/0 \

3:boot uos by pass-through audio in cmd line(HDAC @ 00:17.0 as codec HDAS @0:e:0 as audio device).

4.disassemble the ACPI table and display its contents in uos
cp /sys/firmware/acpi/tables/DSDT . 
iasl -d DSDT  cat DSDT.dsl 

5.Check below entry if exist in uos DSDT.dsl 
    Device (I2C0) Device (HDAS)
root@clr-da2b467367eb45fdbbbdca81b1664b39 / # cat /mnt/DSDT.dsl |grep -e "I2C0"
root@clr-da2b467367eb45fdbbbdca81b1664b39 / # cat /mnt/DSDT.dsl |grep -e "HDAS"
Device (HDAS)
ADBG ("HDAS _INI")
ADBG ("HDAS _DSM")
Return (NBUF) /* _SB_.PCI0.HDAS.NBUF */
Concatenate (Arg0, Arg1, GBUF) /* _SB_.PCI0.HDAS.ACCG.GBUF */
Return (GBUF) /* _SB_.PCI0.HDAS.ACCG.GBUF */

6. ssh to sos and disassemble sos DSDT follow Step4

7. Check below entry if exist in SOS DSDT.dsl
            Device (HDAC)             Device (HDAS) 
And check related I2C entry in DSDT.dsl,  if showed as below, Device(I2C4) should be SOS DSDT.dsl
    Scope (_SB.PCI0.I2C4)

Expected result:
Step 5, Device (I2C0) exists in uos DSDT.dsl 

Actual result:
Step 5, Device (I2C0) not exists in uos DSDT.dsl 

Reproduce rate: 100%

Debugging info:
please see the attachment.

mwang106, 2018-12-06:
[External_System_ID] ACRN-1970

Mingyuan18, 2018-12-24:
QA will delete related cases on APL NUC according to real physical HW, and close this bug.

closed 2018-12-24 by Mingyuan18