projectacrn/acrn-hypervisor #2006: [MRB][IO][AaaG]Report virtio_input_send_event: not enough avail descs... or uos kernel panic, when press the keyboard in virtio mode

jingrongx, 2018-12-07:
Platform/Program:
GP2.0-APL-MRB

Found in build:
Kernel Version/Android Version: 4.19.0-quilt-2e5dc0ac-g1172100caeb9/ 9
AOSP Version: PPR1.181005.003
ABL Version: 1846_GP20
IFWI Version: 3.1.55.2278a
IOC Version: 4.0.12
ACRN-Hypervisor: 2018w48.1.173000p_123
SOS Build Link: 20181128-26540
SOS Version: Installed version: 26540
SOS Kernel Version: 4.19.0-26.iot-lts2018-sos #1 SMP PREEMPT Wed Nov 28
cAVS FW   Version: 9.22.01.3774

Operating System: 
Android as a guest

Test steps:
1. plug in USB keryboard with USB3.0 port and make sure virtio-input enabled in kernel config
CONFIG_VIRTIO_INPUT=y
2.   Remove USB related pass thru/mediator parameters in Launch.sh
3.   check input device info. And find the input event in SOS:
           $cat /proc/bus/input/devices
              $ls -l /dev/input/by-id
4.   add virtio-input parameter in launch.sh:
             -s 7,virtio-input,/dev/input/event2
5.   boot up uos and check keyboard work on uos

Expected Result:
keyboard works normally in uos

Test Result:
Report virtio_input_send_event: not enough avail descs... or uos kernel panic

[input_keyboard_AaaG_dmesg.log](https://github.com/projectacrn/acrn-hypervisor/files/2656248/input_keyboard_AaaG_dmesg.log)
[virtio_input_error.log](https://github.com/projectacrn/acrn-hypervisor/files/2656249/virtio_input_error.log)

please see the attachment.

wenlingz, 2018-12-12:
@jingrongx please add External_System_ID

Mingyuan18, 2018-12-24:
Can't reproduce this issue in below build: 
ACRN: 2018w50.3.140000p_131
SOS : 26800

closed 2018-12-24 by Mingyuan18