projectacrn/acrn-hypervisor #2016: [MRB][HV][AaaG]acrnctl reset fail.

taoyuhong, 2018-12-10:
1.Environment
 --------------------------------------------------------------
 MRB image(SIT):
 >Link : http://sit-image.sh.intel.com/acrn_daily_build/ww47.5/MRB/B1/ 
 >AaaG Image Link:    https://buildbot.sh.intel.com/absp/builders/master-latest/builds/2412
 Installed version: 26400
 Version URL: https://cdn.download.clearlinux.org/update/
 Content URL: https://cdn.download.clearlinux.org/update/ 
 Build Info:
 Linux clr-d62d549093f94ce9ac1a4b639ab448e3 4.19.0-quilt-2e5dc0ac #1 SMP PREEMPT Fri Nov 23 01:08:34 UTC 2018 x86_64 GNU/Linux
 DM version is: 0.4-unstable-9d529fb9 (daily tag:acrn-2018w47.4-140000p), build by root@2018-11-23 01:10:32

2.Reproduce step:
 --------------------------------------------------------------
 1). Boot to SOS
 2). SSH to SOS
 3). Add a Laag UOS into list over SOS SSH by
 acrnctl add [uos_bash_script]
    ex: acrnctl add launch_UOS.sh -V 1
 4). Start that VM by below command
     acrnctl start [vm name]
   Notes: use "acrnctl list" to check the vm name, if you don't know it. 
     ex: acrnctl start vm1-13:ce:dd
 5). Over console reset vm by below command
     acrnctl reset [vm name]
     Notes: use "acrnctl list" to check the vm name, if you don't know it. 

3. Current result:
 --------------------------------------------------------------
 Uos cannot boot up successfully.

4. Expected result:
 --------------------------------------------------------------
 Uos could boot up successfully.

5. repetition rate
 --------------------------------------------------------------
 100%

6.Logs:
 --------------------------------------------------------------
 root@clr-d62d549093f94ce9ac1a4b639ab448e3 ~ # acrnctl reset vm1-14:5b:5a
 Failed to stop vm1-14:5b:5a in 10 sec

[External_System_ID] ACRN-1922

mwang106, 2018-12-17:
[External_System_ID] ACRN-1922

Mingyuan18, 2019-01-02:
Can't reproduce this issue in below build: 
ACRN: 2018w51.2.140000p_134; SOS: 26830

closed 2019-01-02 by Mingyuan18