projectacrn/acrn-hypervisor #2031: UP2 serial port has no output with uart=mmio@0x91622000 for hypervisor

mwang106, 2018-12-11:
1.Environment
[Board]: APL UP2
daily build：https://download.clearlinux.org/releases/26740/ 
Image info：26740

2.Steps
Replace acrn.efi After modifying acrn.conf, execute "efibootmgr -c -l "\EFI\acrn\acrn.efi" -d /dev/mmcblk0 -p 1 -L "ACRN Hypervisor NUC" -u "bootloader=\EFI After \org.clearlinux\bootloaderx64.efi uart=mmio@0x91622000"", After the SOS is started, the screen is not displayed, the ssh connection can be made, and the serial port is not output.

3. Current result: SOS boot screen no display serial port no output
4. Expected result: The SOS starts successfully, the screen display is normal, and the serial port output is normal.

mwang106, 2018-12-11:
[External_System_ID] ACRN-2238

Mingyuan18, 2019-01-09:
Cannot be reproduced in this build:
Installed version: 27050

closed 2019-01-09 by Mingyuan18