projectacrn/acrn-hypervisor #2036: Update and complete `acrn-dm` parameters description in the user guide and HLD

gvancuts, 2018-12-12:
The `acrn-dm` user guide (PR #2022) describes the parameters that `acrn-dm` can take. Based on the output of `acrn-dm -v`, we are missing a description for some of them, specifically:
```
       -E: elf image path
       --part_info: guest partition info file path
       --debugexit: enable debug exit function
       --intr_monitor: enable interrupt storm monitor
       --vtpm2: Virtual TPM2 args: sock_path=$PATH_OF_SWTPM_SOCKET
............its params: threshold/s,probe-period(s),delay_time(ms),delay_duration(ms)
```

There are three small updates this issue intends to track:
1. Update the user guide with the missing parameter descriptions
2. Update the HLD document with the latest `acrn-dm -v` output
3. Update the code to clean the output of `vtpm2`, remove the series of dot on the second line

gvancuts, 2018-12-12:
@NanlinXie @dbkinder @hwang37

Mingyuan18, 2018-12-21:
[External_System_ID] ACRN-2294

Mingyuan18, 2019-01-11:
ACRN DM document updated.

closed 2019-01-11 by Mingyuan18