projectacrn/acrn-hypervisor #2076: The acrn hypervisor of release version is not support 32G memory

hongww0425, 2018-12-17:
The acrn hypervisor in Clear Linux version 26200 or newer.When the memory is 32G in nuc devices,loading acrn.efi failed.! There are no log print in console, the screen is black and the system is freeze.
However,when memory is 16G,loading acrn.efi success! There are logs print in console and the system started successfully.

hongww0425, 2018-12-17:
Try to modify 4 lines of content in the file(arch/x86/Kconfig), modify default value from 16G to 32G,and add 64M size for acrn hypervisor. the acrn.efi is loading success and the system started successfully.

mwang106, 2018-12-19:
Will add this kind of configuration information in acrn open source document.

closed 2018-12-19 by mwang106

gvancuts, 2019-01-06:
> Will add this kind of configuration information in acrn open source document.

Have we documented this anywhere yet? If not, could we at least post the detailed info in this issue?