projectacrn/acrn-hypervisor #2156: Enhance the "Building from source" documentation

gvancuts, 2018-12-21:
We should provide more details (including workflows) on how to configure and build ACRN. Specifically, we should provide additional details on how to use new `Makefile` variables such as: `BOARD`, `FIRMWARE`, and mention `PLATFORM` is now obsolete and should not be used anymore.

As part of this enhancement, we should document how to use the hypervisor `Kconfig` system when building from the top-level folder (as building directly from within subfolders is not supported).

yifanddd, 2019-04-12:
[External_System_ID] ACRN-3377

fuzhongl, 2019-05-14:
Hi Geoffroy,
Do you still have more concern for this issue? Or this issue can be closed?
Thanks!

gvancuts, 2019-05-14:
> Do you still have more concern for this issue? Or this issue can be closed?

I still have concerns about our instructions, specifically on how to properly use `Kconfig` to change the default configuration. Here is an example of why I'm concerned:
* Clone the repo and build the entire project:
```
cd acrn-hypervisor
make BOARD=nuc6cayh
```
And you will get an `acrn.efi` file under `build/hypervisor`

Now, do the following:
```
cd acrn-hypervisor/hypervisor
make defconfig BOARD=nuc6cayh
make
```
And you don't get an `acrn.efi` file as output... (under `build/`)

Another concern I have is that if you run `make BOARD=nuc7i7bnh` from the top-level folder, you don't even get an `acrn.efi` file in that case.

fuzhongl, 2019-07-22:
> Now, do the following:
> 
> ```
> cd acrn-hypervisor/hypervisor
> make defconfig BOARD=nuc6cayh
> make
> ```
> 
> And you don't get an `acrn.efi` file as output... (under `build/`)

@gvancuts To get an `acrn.efi` file; not need cd acrn-hypervisor/hypervisor 
Please do it at  the top-level folder

> Another concern I have is that if you run `make BOARD=nuc7i7bnh` from the top-level folder, you don't even get an `acrn.efi` file in that case.

@gvancuts please using following command:
`make BOARD=nuc7i7bnh FIRMWARE =uefi'

BTW: we will restructure the make file for ARM support.

fuzhongl, 2019-07-22:
> I still have concerns about our instructions, specifically on how to properly use `Kconfig` to change the default configuration. Here is an example of why I'm concerned:
>
> * Clone the repo and build [...]

closed 2019-07-22 by fuzhongl