projectacrn/acrn-hypervisor #2170: For UEFI based hardware platforms, one ClearLinux E2E build binary can be used for all platform's installation

mwang106, 2018-12-25:
The E2E build binary should include ACRN component, vSBL, SOS, LaaG image.
The E2E build binary should support IVI usage by sharing mode, Real-time usage by partition mode.
The open source platforms to be supported: Apollo Lake NUC, Kabylake NUC, Apollo Lake UP2, with UEFI firmware as boot loader.
 ACRN hypervisor only need one UEFI kconfig file for all. In the long run, should remove different configure file for different hardware platform
ClinuxLinux E2E GSG may have addition steps to guide community developer on how to boot E2E build on different platforms, by changing efibootmgr params for different hardware platforms.

mwang106, 2018-12-25:
[External_System_ID] ACRN-2274

Mingyuan18, 2019-01-10:
Not found any Regression issue in test. So clost it

closed 2019-01-10 by Mingyuan18