projectacrn/acrn-hypervisor #2191: Kernel panic while run adb reboot with SOS run daimler ic.

zhipenggong, 2018-12-26:
[2018-12-26 11:43:46] [20082.307198] BUG: unable to handle kernel NULL pointer dereference at 00000000000002bc
[2018-12-26 11:43:46] [20082.310230] PGD 0 P4D 0 
[2018-12-26 11:43:46] [20082.312874] Oops: 0000 [#1] PREEMPT SMP
[2018-12-26 11:43:46] [20082.315584] CPU: 0 PID: 485 Comm: acrngt_emulatio Tainted: G     U  W         4.19.8-quilt-2e5dc0ac-00005-g100be0b #4
[2018-12-26 11:43:46] [20082.318744] RIP: 0010:shadow_ppgtt_mm+0x3d5/0x420
[2018-12-26 11:43:46] [20082.321534] Code: b6 e8 d2 e3 9d ff eb de 48 c7 c6 b0 ca 76 b6 48 c7 c7 e1 b1 79 b6 e8 1a f8 97 ff 0f 0b b8 ea ff ff ff e9 5d fd ff ff 4d 85 ff <48> 8b 90 c8 02 00 00 74 26 49 81 ff 00 f0 ff ff 77 1d 41 8b 77 48
[2018-12-26 11:43:46] [20082.325353] RSP: 0018:ffff951d8faf3c90 EFLAGS: 00010286
[2018-12-26 11:43:46] [20082.397249] RAX: fffffffffffffff4 RBX: ffff951da180a540 RCX: ffffd157896a7307
[2018-12-26 11:43:46] [20082.400387] RDX: 0000000000000004 RSI: ffffffffb565c878 RDI: 0000000000000286
[2018-12-26 11:43:46] [20082.403535] RBP: ffff951d8faf3d00 R08: 0000000000000000 R09: 0000000000aaaaaa
[2018-12-26 11:43:46] [20082.406688] R10: 0000000000000000 R11: 00000000ffffffff R12: ffff951da180a540
[2018-12-26 11:43:46] [20082.442380] R13: ffff951c58e78838 R14: ffff951da180a5b8 R15: ffffadb242239000
[2018-12-26 11:43:46] [20082.445562] FS:  0000000000000000(0000) GS:ffff951db3a00000(0000) knlGS:0000000000000000
[2018-12-26 11:43:46] [20082.466124] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[2018-12-26 11:43:46] [20082.469259] CR2: 00000000000002bc CR3: 0000000265b7c000 CR4: 00000000003406f0
[2018-12-26 11:43:46] [20082.472517] Call Trace:
[2018-12-26 11:43:46] [20082.475485]  ? kmem_cache_alloc_trace+0x1f1/0x260
[2018-12-26 11:43:46] [20082.478604]  ? __switch_to_asm+0x40/0x70
[2018-12-26 11:43:46] [20082.481669]  ? __switch_to_asm+0x34/0x70
[2018-12-26 11:43:46] [20082.484723]  intel_vgpu_create_ppgtt_mm+0xa0/0x1b0
[2018-12-26 11:43:46] [20082.525290]  intel_vgpu_get_ppgtt_mm+0x4d/0x90
[2018-12-26 11:43:46] [20082.533411]  pvinfo_mmio_write+0x222/0x2d0
[2018-12-26 11:43:46] [20082.536491]  ? _raw_spin_unlock_irqrestore+0x43/0x60
[2018-12-26 11:43:46] [20082.539630]  ? finish_wait+0x72/0x90
[2018-12-26 11:43:46] [20082.542678]  ? trace_hardirqs_on+0x38/0x100
[2018-12-26 11:43:46] [20082.545759]  intel_vgpu_mmio_reg_rw+0x190/0x4c0
[2018-12-26 11:43:46] [20082.548870]  ? __might_sleep+0x4a/0x80
[2018-12-26 11:43:46] [20082.551928]  intel_vgpu_emulate_mmio_write+0x13d/0x2b0
[2018-12-26 11:43:46] [20082.555085]  acrngt_emulation_thread+0x2c8/0x400
[2018-12-26 11:43:46] [20082.558202]  kthread+0x12c/0x150
[2018-12-26 11:43:46] [20082.561221]  ? acrngt_dma_unmap_guest_page+0x10/0x10
[2018-12-26 11:43:46] [20082.564360]  ? kthread_create_worker_on_cpu+0x70/0x70
[2018-12-26 11:43:46] [20082.567508]  ret_from_fork+0x3a/0x50

Mingyuan18, 2019-01-11:
[External_System_ID] ACRN-2263

Mingyuan18, 2019-01-21:
Can't reproduce this issue on below build,  so close this issue.
ACRN-Hypervisor: 2019w02.2.150000p-141
SOS : 27151

closed 2019-01-21 by Mingyuan18