projectacrn/acrn-hypervisor #2200: Won't build using ubuntu 16.04 LTS and binutils 2.26.1

Vshmuk, 2018-12-26:
The file `arch/x86/idt.S` includes `include/arch/x86/idt.h`. The following piece of code:

`#define     X64_IDT_DESC_SIZE   (0x10U)

/* Number of the HOST IDT entries */

#define     HOST_IDT_ENTRIES    (0x100U)

/* Size of the IDT */

#define     HOST_IDT_SIZE       (HOST_IDT_ENTRIES*X64_IDT_DESC_SIZE)

`

is expanded to 

`
.short  ((0x100U)*(0x10U)) - 1
`,

which is not recognized by `as` as correct code. The output of assembler is:

`
arch/x86/idt.S: Assembler messages:

arch/x86/idt.S:19: Error: missing ')'

arch/x86/idt.S:19: Error: missing ')'

arch/x86/idt.S:19: Error: junk at end of line, first unrecognized character is `U'
`

`
$ as -v

GNU assembler version 2.26.1 (x86_64-linux-gnu) using BFD version (GNU Binutils for Ubuntu) 2.26.1
`

Mingyuan18, 2019-01-10:
[External_System_ID] ACRN-2434

Mingyuan18, 2019-01-30:
@Vshmuk , Please help to confirm whether this bug has been solved? If yes, please close it. Thanks!

Mingyuan18, 2019-01-31:
Verify pass on acrn-tag:  acrn-2019w05.2-140000p .

closed 2019-01-31 by Mingyuan18

Vshmuk, 2019-01-31:
Now it says:

/usr/bin/ld: warning: -z noreloc-overflow ignored.
/usr/bin/ld: /home/xxx/projects/acrn-hypervisor/build/hypervisor/boot/reloc.o: relocation R_X86_64_32S against `.text.get_hv_image_delta' can not be used when making a shared object; recompile with -fPIC
/home/ebeysembaev/projects/acrn-hypervisor/build/hypervisor/boot/reloc.o: error adding symbols: Bad value
collect2: error: ld returned 1 exit status

Something is broken in my setup?

gvancuts, 2019-04-24:
You're hitting https://github.com/projectacrn/acrn-hypervisor/issues/2200. It was closed based on the report of users that the problem had been solved, but I'm hitting it too now using the latest `binutils` from the Ubuntu 16.04 repository (i.e. Version: 2.26.1-1ubuntu1~16.04.8).

I'll try to upgrade my `binutils` to a newer version to see if it solves the issue.