projectacrn/acrn-hypervisor #2227: 1080p60fps Video playback performance failed to reach target 60 fps, but only to 30 fps.

dandanyx, 2019-01-02:
Issue Summary:
-------------------------
In AaaG-Acrn, 
$adb push 1080p60fps.m4 /sdcard/Movies

By Android Gallery tool, playback this media.
It only reach about 30 fps, not expected 60 fps.

Issue Description:
log filenames + line numbers showing issue:
MA/Artemis logs
LTE (istp) logs
Crash ID/Call Stack

Testcase ID:
-------------------

Expected test result:
--------------------------------
Testcase Description/Procedure/mobility or stationary?

Actual test result:
---------------------------

Additional Information:
-------------------------------------

Test report Date: WKxx, dd.mm.yyyy
NW Operator: (including SW/HW Revision if possible)
Test Location: (city, country)
Test Campaign: (i.e. ADAPT, GCF, KPI, DG11, etc)
SW/HW Rev Version: DEV_XMM7260_MODEM_01.yyww.mm / XMM7260_REV_x.0
SW path location:
HW ID: i.e. CN200.1.x.y
UE Reference: i.e. Samsung Infuse 4G, Motorola Atrix 4G
UE Reference test result:

dandanyx, 2019-01-02:
[External_System_ID] OAM-72444

Mingyuan18, 2019-03-22:
Can't reproduce this issue in below build:
ACRN-Hypervisor : 2019w05.1.150000p_155
SOS: 27217

closed 2019-03-22 by Mingyuan18