projectacrn/acrn-hypervisor #2237: Don't export two dma_bufs for the same importer in sos kernel

yakuizhao, 2019-01-07:
After receiving the info that one new importer is sent from UOS,  it is possible that the EXPORT_FD_IOCTL is called twice and then each dma_buf is created for one EXPORT_FD_IOCTL. 
As the dma_buf has the refcnt,  it is unnecessary to export two dma_bufs for the same importer. 

At the same time in course of guest reboot, the UOS FE driver has no chance to send the UNEXPORTED_FD msg. In such case the importer info in SOS can't be released in time.

yakuizhao, 2019-01-07:
[External_System_ID] ACRN-2411

yakuizhao, 2019-01-22:
Now the  patches are already merged to 4.19/base branch of github.com/intel-linux-lts.git.
So this issue will be closed.

Mingyuan18, 2019-01-22:
No regression found in test.

closed 2019-01-22 by Mingyuan18