projectacrn/acrn-hypervisor #2251: ipanic hit during AaaG warm reboot testing: hsw_power_well_enable

dandanyx, 2019-01-08:
Issue Summary: Met IPANIC DATA1=khungtaskd DATA2=hung_task  during warm boot testing   

Frequency: 2/10 DUTs

Preconditons:
1.Wired network support
2.Setup IPATH test env

(1) Download the IPATH installer:
(2) Execute the following command to install IPATH:
python IPATH_installer.pyc

Steps to Reproduce:
1. Connect the device with debug card to test host by USB, start minicom log
2. To IPATH test env, modify test campaign and configure file, and start test
(1) cd to IPATH/testsuite/campaign/kpi$ vi campaign_kpi_BXTP_P_ACRN_WB.xml
(2) cd to IPATH/config/ vim x_device1_execution.config & x_Device_Config_1.py
(3)cd to IPATH/execute, run with commend:
python execute_tests.py ../config/R1J56La006ab82_device1_execution.config
3. Check the test status during testing

Expect result:
All case test normally,no critical crash happened.

Actual result:
Met IPANIC DATA1=khungtaskd DATA2=hung_task  during warm boot testing

dandanyx, 2019-01-08:
[External_System_ID] ACRN-2418,OAM-73767

Mingyuan18, 2019-03-21:
Not reproduced this issue in below build:
ACRN-Hypervisor : 2019w03.5.150000p_148
SOS : 27215

closed 2019-03-21 by Mingyuan18