projectacrn/acrn-hypervisor #2267: [APLUP2][LaaG]LaaG can't detect 4k monitor

xiaoxiax, 2019-01-09:
Environment:
------------
HSW: APLUP2

Bug detailed description:
-------------
LaaG can't detect 4k monitor,4k monitor does not display.

Reproduce steps:
-------------

1.Connnet 4k monitor.
2. Boot up hypervisor and SOS and UOS

Current result:
--------------
LaaG can't detect 4k monitor,4k monitor does not display.

Expected result:
--------------
LaaG can detect 4k monitor,4k monitor can display.

xiaoxiax, 2019-01-09:
[External_System_ID] ACRN-2423

mwang106, 2019-01-16:
Mark this issue as a known limitation on APL platform.

closed 2019-01-16 by mwang106

gvancuts, 2019-01-16:
> Mark this issue as a known limitation on APL platform.

Can you explain in a few words where the limitation comes from?