projectacrn/acrn-hypervisor #2279: Mediator Usb_KeyBoard and Mouse and Exit UOS, SOS can not use the Usb_KeyBoard and Mouse

Mingyuan18, 2019-01-10:
Environment:
HW/Board：APL NUC
Image info:
IOC version: NUC has not support IOC.
SOS version:26950
SOS kernel version:Linux sos-dom0 4.19.8-quilt-2e5dc0ac #1 SMP PREEMPT Wed Jan 2 05:21:01 UTC 2019 x86_64 GNU/Linux
UOS Type: LaaG
UOS version: 26950
UOS kernel version:Linux clr-8ffd088f8e3e4f9195996c2bb7c612e2 4.19.8-quilt-2e5dc0ac #1 SMP PREEMPT Wed Jan 2 05:22:48 UTC 2019 x86_64 GNU/Linux
steps:
1:insert USB keyboard and mouse in standard A port（USB3.0 port）
root@sos-dom0 ~ # ls /dev/input/by-id
usb-Logitech_USB_Receiver-event-if01 usb-Logitech_USB_Receiver-if01-event-mouse
usb-Logitech_USB_Receiver-event-kbd usb-Logitech_USB_Receiver-if01-mouse
2:boot uos by sharing the USB keyboard and mouse in cmd line 
-s n,xhci,1-1:1-2:1-3:1-4:2-1:2-2:2-3:2-4 \
3:uos can access USB keyboard and mouse.
root@clr-8ffd088f8e3e4f9195996c2bb7c612e2 ~ # ls /dev/input/by-id/
usb-Logitech_USB_Receiver-event-if01 usb-Logitech_USB_Receiver-if01-event-mouse
usb-Logitech_USB_Receiver-event-kbd usb-Logitech_USB_Receiver-if01-mouse
4:exit uos.
root@clr-8ffd088f8e3e4f9195996c2bb7c612e2 ~ # poweroff
5:sos can access USB keyboard and mouse.
Expected result:
Step 5, sos can access USB keyboard and mouse
Actual result:
Step 4, sos can not access USB keyboard and mouse,must hot plugin the USB keyboard and mouse.
Reproduce rate: 100%

Mingyuan18, 2019-01-10:
[External_System_ID] ACRN-2404

arun1778, 2019-05-04:
is this issue still open or fix available for same?

Mingyuan18, 2019-06-05:
Since not reproduce on KBL NUC, close it with "won't fix".

closed 2019-06-05 by Mingyuan18