projectacrn/acrn-hypervisor #2347: [tools] Don't hardcode install paths

rossburton, 2019-01-15:
The tools hardcode install locations, which is bad practise as not every distro is the same.

Would it be better to just replace the Makefiles with Meson or similar?

gvancuts, 2019-01-15:
> Would it be better to just replace the Makefiles with Meson or similar?

I'm all in favour of that, +1!

Mingyuan18, 2019-01-18:
[External_System_ID] ACRN-2539

lyan3, 2019-04-24:
It's a 

> The tools hardcode install locations, which is bad practise as not every distro is the same.
> 
> Would it be better to just replace the Makefiles with Meson or similar?

Thanks for the good suggestion. I will have it as one future work for now, and of course, it needs to be resolved before more distro other than ClearLinux and Ubuntu is supported as SOS.

fuzhongl, 2019-05-30:
@rossburton 
Thanks for reporting this issue; we are working on this issue and will keep you update in future.

rossburton, 2019-06-02:
How are you working on it?  If you're making the paths variables in the makefiles then I have a patch you can start with.  If you're moving to Meson then I'll help where I can.

closed 2020-04-21 by fuzhongl