projectacrn/acrn-hypervisor #2371: kconfig oldconfig doesn't work correctly

rossburton, 2019-01-17:
As I understand it, I should be able to write a .config programatically like this:

```
CONFIG_BOARD="nuc6cayh"
CONFIG_RELEASE=y
CONFIG_FIRMWARE_UEFI=y
```

And running `olddefconfig` should fill in the blanks, using those values instead of the defaults.  This doesn't work, for some reason the inclusion of the board-specific pieces never happens.

gvancuts, 2019-01-17:
That's an interesting question, I don't think we have ever tested using `olddefconfig`. We typically create a board-specific config and use it by calling `make defconfig BOARD=xxx` (and the config file should be called `xxx.config`).

There is some redundancy in how we define the `BOARD` and `FIRMWARE` variables, and unfortunately no cross-check so you can fairly easily end up compiling something you did not mean to. This is something we ought to fix at some point.

Mingyuan18, 2019-01-22:
[External_System_ID] ACRN-2548

junjiemao1, 2019-01-22:
@rossburton Thanks for reporting this. Unfortunately `olddefconfig` is not yet supported. The functionality you would like to have should be part of `oldconfig` which is similar to `silentoldconfig` in Linux, but unfortunately `oldconfig` does not work in this case either because the build scripts does not update .config when there is any symbol missing.

I have drafted a patch #2388 to fix the behavior of `oldconfig` so that it works similar to what you expect from `olddefconfig` in your case. Please kindly let me know if it meets your needs. Thanks.

junjiemao1, 2019-01-22:
Please also take care when editing `.config`, especially the CONFIG_BOARD symbol which is compared against those in arch/x86/configs, case sensitively.

gvancuts, 2019-01-22:
> @gvancuts I was wondering if `BOARD` and `FIRMWARE` duplicate each other. Even on the same `BOARD` the hypervisor may still run on different `FIRMWARE`, either on UEFI directly or on a multiboot-compliant chained loader. That's why we split the deprecate the previous `PLATFORM` and replace it with `BOARD` + `FIRMWARE`.

There is some overlap in my opinion between the two and I think we should clean it (also because there is no cross-check done to ensure consistency at build time). What I think we should do is:
* Use `BOARD` as the **only** variable we pass on to `make`. The `BOARD=` parameter points at a config file which includes all configuration options, **including** the firmware that is used.
* Remove `CONFIG_BOARD` from Kconfig (it's useless in there, you can change it from within the `make menuconfig` but that has no effect).
* Have every configuration have its own dedicated config file, so in the case of the UP2 board for example, there would be two config files: `up2-uefi.config` and `up2-sbl.config`

We should maintain a subset of config files that we test with, we cannot cover all configurations. ACRN users will most likely end up defining their own config file which matches their platform, use-case and global system configuration.

rossburton, 2019-01-22:
I was wondering if it would be possible to remove both BOARD and FIRMWARE from the makefile environment entirely.  For the hypervisor itself require that a config file exists, and that can have all the configuration in.  The makefile can use the values directly as per https://github.com/ulfalizer/Kconfiglib#using-config-files-as-make-input.

gvancuts, 2019-01-22:
> I was wondering if it would be possible to remove both BOARD and FIRMWARE from the makefile environment entirely. For the hypervisor itself require that a config file exists, and that can have all [...]

I must confess I'm not understanding this completely. Can you give an example of how you would compile the hypervisor for a specific config? I'm assuming this then involves some direct file manipulations such as copying `xxx.config` to `.config` or am I missing something?

totoroow, 2019-01-23:
Hi @gvancuts 

> Use BOARD as the only variable we pass on to make. The BOARD= parameter points at a config file which includes all configuration options, including the firmware that is used.

I think `BOARD` may not be enough. As we now support one board with two different firmwares(SBL or UEFI), that's the reason why we pass `BOARD` and `FIRMWARE` to `make` when build hypervisor.

gvancuts, 2019-01-23:
> Hi @gvancuts
>
>
> I think `BOARD` may not be enough. As we now support one board with two different firmwares(SBL or UEFI), that's the reason why we pass `BOARD` and `FIRMWARE` to `make` when build [...]

Just keep two separate config files just like I mentionned, e.g. `up2-uefi.config` and `up2-sbl.config`. There could also be different settings we want for these different boot firmware.

totoroow, 2019-01-23:
> Just keep two separate config files just like I mentionned, e.g. up2-uefi.config and up2-sbl.config. There could also be different settings we want for these different boot firmware.

I mean the parameters we pass on to make. I'm fine with two config files. Just a little concern that these two files are almost same except `CONFIG_PLATFORM_[SBL| UEFI]`.

gvancuts, 2019-01-23:
> I mean the parameters we pass on to make. I'm fine with two config files. Just a little concern that these two files are almost same except `CONFIG_PLATFORM_[SBL| UEFI]`.

There are indeed very similar today, except for the firmware. We don't have to keep it that way though. We could decide that we maintain two very different config files to show the users a more advanced use-case. We could introduce additional differences in terms of how we allocate the RAm memory and use very different log levels for each.

My concern is that we duplicated a configuration variable by defining a new one that can be used from the command-line, just so we can maintain one less example config file. This is a lot of extra complexity on the Kconfig/Makefile logic just to avoid adding one file.

On my personal dev machine, I already maintain different config files for the very same board. I have one for example where I use a different UEFI bootloader (when I use Ubuntu as the Service OS). I have another one for the NUC7i7BNH (which does not have a serial port), etc. This mechanism using config files is in place so that you can easily and centrally maintain the entire configuration of your system(s), duplicating some select config options to be able to use them from the command-line goes against the principle we followed when we introduced Kconfig.

totoroow, 2019-01-24:
Hi @gvancuts  I understand.

How do you think about `BOARD=up2-sbl` versus `BOARD=up2 FIRMWARE=sbl` ?
For now we use the latter one which is more clear. 
I think we could use the combination of board and firmware (such as up2-sbl for example) as the name of the config. I think this is what you mean.

junjiemao1, 2019-01-24:
@rossburton Would you please confirm if the reported issue has been resolved? If so how about we close this one and create a new thread for the refinement of `BOARD` + `FIRMWARE`?

gvancuts, 2019-01-24:
> Hi @gvancuts I understand.
> 
> How do you think about `BOARD=up2-sbl` versus `BOARD=up2 FIRMWARE=sbl` ?
> For now we use the latter one which is more clear.
> I think we could use the combination of board and firmware (such as up2-sbl for example) as the name of the config. I think this is what you mean.

Yes, that's what I mean.

gvancuts, 2019-01-24:
> @gvancuts `CONFIG_BOARD` is there because we need handle cases like invoking `make BOARD=up2` while there is an existing .config for `mrb`. This symbol is used to determine if the two `BOARD` from command line and existing .config are the same. In addition, `CONFIG_BOARD` is also used to determine the proper defconfig. Refer to hypervisor/arch/x86/Kconfig for the detailed definition of `DEFCONFIG_LIST`.

I _think_  I see what you mean to some extent. I was playing with that yesterday trying to get rid of `CONFIG_BOARD` in the `Kconfig` file but I was running into some issues (that I did not get to the bottom of yet).

I'm never 100% sure how `kconfiglib` commands should behave. One thing for example is that we have different default values for some config options that depend on whether we use `SBL` or `UEFI`. If you start from a default config for example, run `make menuconfig` and change from `SBL` to `UEFI`, that does not update the other config options. Is this expected?

Mingyuan18, 2019-01-29:
@rossburton Would you please confirm if the reported issue has been resolved?

rossburton, 2019-01-29:
Confirmed.

closed 2019-01-29 by rossburton

rossburton, 2019-02-15:
This isn't quite working as I was hoping actually:

```
echo CONFIG_BOARD=\"nuc6cayh\" >${B}/hypervisor/.config
echo CONFIG_UEFI_OS_LOADER_NAME=\"FOO/BAR\" >>${B}/hypervisor/.config
make -C hypervisor oldconfig BOARD= nuc6cayh FIRMWARE=uefi
```

Results in:

make: Entering directory '/data/poky-tmp/master/work/intel_corei7_64-poky-linux/acrn-hypervisor/0.5+2019w05.4-r0/git/hypervisor'
Overwrite with default configuration based on arch/x86/configs/nuc6cayh.config.
Configuration written to /data/poky-tmp/master/work/intel_corei7_64-poky-linux/acrn-hypervisor/0.5+2019w05.4-r0/build//hypervisor/.config.

And UEFI_OS_LOADER_NAME being overwritten.

junjiemao1, 2019-02-15:
@rossburton There are two places that need be tweaked in the command. The first is the board name. Currently we use upper case for `CONFIG_BOARD` and this symbol is case-sensitive, which means the following command should be used when manually editing `.config`.
```
echo CONFIG_BOARD=\"NUC6CAYH\" >${B}/hypervisor/.config
```
The one provided to `BOARD=` on the command line can be either upper-case or lower-case, though. 

The second place is that you need to append a `HV_OBJDIR` to specify the build directory if the default doesn't work for you. From your commands it seems that you are working at the top-level directory of the repo and would like to use `${B}/hypervisor` as the build directory. The following command should work in this case, provided that `${B}` is an absolute path.
```
make -C hypervisor oldconfig HV_OBJDIR=${B}/hypervisor BOARD=nuc6cayh FIRMWARE=uefi
```

The following commands work properly on my box.
```
~/testbed/acrn$ mkdir -p build/hypervisor
~/testbed/acrn$ echo CONFIG_BOARD=\"NUC6CAYH\" > build/hypervisor/.config
~/testbed/acrn$ echo CONFIG_UEFI_OS_LOADER_NAME=\"FOO/BAR\" >> build/hypervisor/.config
~/testbed/acrn$ cat build/hypervisor/.config
CONFIG_BOARD="NUC6CAYH"
CONFIG_UEFI_OS_LOADER_NAME="FOO/BAR"
~/testbed/acrn$ make -C hypervisor HV_OBJDIR=`pwd`/build/hypervisor oldconfig BOARD=nuc6yach FIRMWARE=uefi
make: Entering directory `/home/junjie/testbed/acrn/hypervisor'
Configuration written to /home/junjie/testbed/acrn/build/hypervisor/.config.
make: Leaving directory `/home/junjie/testbed/acrn/hypervisor'
make: Entering directory `/home/junjie/testbed/acrn/hypervisor'
make: Nothing to be done for `oldconfig'.
make: Leaving directory `/home/junjie/testbed/acrn/hypervisor'
~/testbed/acrn$ grep "UEFI_OS_LOADER" build/hypervisor/.config
CONFIG_UEFI_OS_LOADER_NAME="FOO/BAR"
```

rossburton, 2019-02-15:
> The one provided to `BOARD=` on the command line can be either upper-case or lower-case, though.

That's not true, CONFIG_BOARD needs to be uppercase and BOARD needs to be lowercase.

rossburton, 2019-02-15:
Okay, looks like I made it work. Thanks!

gvancuts, 2019-02-15:
> Okay, looks like I made it work. Thanks!

Can you share your exact sequence of instructions?

rossburton, 2019-02-15:
Pretty much what @junjiemao1 said, I was failing to escape enough.