projectacrn/acrn-hypervisor #2378: Getting started guide for NUC is not working

LibertadCruz, 2019-01-18:
I am following the getting started guide to setup ACRN for Intel on NUC. 

https://projectacrn.github.io/latest/getting-started/apl-nuc.html 

Executing step 6 : Reboot and select “The ACRN Service OS” to boot.  The NUC does not boot as expected it will stay on the welcoming page that says...  INTEL NUC......

My guess is that the configuration options for the kernel variables a re missing something or need to be change in some way.

Please let me know.

ttzeng, 2019-01-19:
@LibertadCruz Please provide detailed context such as the platform/processor you are using, and exactly message shown on screen? I can tell the ACRN v0.5 works on ApolloLake based NUC6CAYH and KebyLake based NUC7i5BNH, FYR.

LibertadCruz, 2019-01-21:
Thanks for the quick response, you will see the information below. After finishing the configuration of the clr-boot-manager I reboot and my NUC gets stuck on the image I uploaded.

$ sudo clr-boot-manager set-timeout 20
$ sudo clr-boot-manager update

 
[image]

System Information
	Manufacturer: Intel corporation
	Product Name: NUC6CAYS
	Version:                                  
	Serial Number: G6AY703008UY
	UUID: 63fc2e61-6bc1-28ae-7867-f44d306ac201
	Wake-up Type: Power Switch
	SKU Number:                                  
	Family: BOXNUC6CAYSAJR

Processor Information
	Socket Designation: SOCKET 0
	Type: Central Processor
	Family: Celeron
	Manufacturer: Intel
	ID: C9 06 05 00 FF FB EB BF
	Signature: Type 0, Family 6, Model 92, Stepping 9
	Flags:
		FPU (Floating-point unit on-chip)
		VME (Virtual mode extension)
		DE (Debugging extension)
		PSE (Page size extension)
		TSC (Time stamp counter)
		MSR (Model specific registers)
		PAE (Physical address extension)
		MCE (Machine check exception)
		CX8 (CMPXCHG8 instruction supported)
		APIC (On-chip APIC hardware supported)
		SEP (Fast system call)
		MTRR (Memory type range registers)
		PGE (Page global enable)
		MCA (Machine check architecture)
		CMOV (Conditional move instruction supported)
		PAT (Page attribute table)
		PSE-36 (36-bit page size extension)
		CLFSH (CLFLUSH instruction supported)
		DS (Debug store)
		ACPI (ACPI supported)
		MMX (MMX technology supported)
		FXSR (FXSAVE and FXSTOR instructions supported)
		SSE (Streaming SIMD extensions)
		SSE2 (Streaming SIMD extensions 2)
		SS (Self-snoop)
		HTT (Multi-threading)
		TM (Thermal monitor supported)
		PBE (Pending break enabled)
	Version: Intel(R) Celeron(R) CPU J3455 @ 1.50GHz
	Voltage: 1.2 V
	External Clock: 100 MHz
	Max Speed: 1500 MHz
	Current Speed: 1500 MHz
	Status: Populated, Enabled
	Upgrade: Other
	L1 Cache Handle: 0x0032
	L2 Cache Handle: 0x0033
	L3 Cache Handle: Not Provided
	Serial Number: Not Specified
	Asset Tag: Fill By OEM
	Part Number: Fill By OEM
	Core Count: 4
	Core Enabled: 4
	Thread Count: 4
	Characteristics:
		64-bit capable

NanlinXie, 2019-01-22:
@LibertadCruz Could you go to native CL to check the CL release version?  CL version 27230 is OK for APL NUC and KBL NUC boot up.

gvancuts, 2019-01-22:
@LibertadCruz please let us know which bios version you are using on the NUC (we've had problems with old-ish versions)

LibertadCruz, 2019-01-24:
thanks for the response: I was using BIOS version: AYAPLCEL.86A.0047.2018.0108.1419 and I updated to the newes version AYAPLCEL.86A.0059. After the update of the BIOS the same thing happens, my NUC does not start with any boot device. Unles I press F10 and choose the built in EFI manually. Then it will give me a nice menu where ACRN is at the top followed by the other kernels. If I select ACRN the screen will go black, no video output. 

If instead of the built in EFI I choose to boot with the ACRN boot device the screen will go straight to black with no video output.

and my clear version is 27320.

ttzeng, 2019-01-24:
@LibertadCruz I met the same issue with CL27230 if I installed my NUC with **Automatic Installation**.  By disabling the "Auto Update" from the Advanced Tab of the installer, I re-installed my NUC with exactly CL27230, then it boots. I suspect there may have regression on KBL NUC after CL27230, so I'd suggest you re-install your KBL NUC using the manual install approach and fix the Clear Linux version at 27230.

gvancuts, 2019-01-24:
> @LibertadCruz I met the same issue with CL27230 if I installed my NUC with **Automatic Installation**. By disabling the "Auto Update" from the Advanced Tab of the installer, I re-installed my NUC [...]

Do we have a bug to track this regression with the upstream project?

LibertadCruz, 2019-01-24:
Ok,  so the results of the following scenarios:

Clear Linux at 27320 + ACRN version corresponding to 27320 clear linux release + Bios at 0047 -->Failed
Clear Linux at 27320 + ACRN version corresponding to 27320 clear linux release + Bios at 0049 -->Failed

Now I will do the following scenarios, maybe they tell us something:

With manual installation + BIOS at the latest version 0059 
1. Clear Linux 27230  +  old ACRN version corresponding to the 27230 clear linux release.
2. Clear Linux latest version +  old ACRN version corresponding to the 27230 clear linux release.
3. Clear linux 27230 +ACRN at the actual version

Unless you see fit another scenario or maybe you would like logs, but I dont think we can have logs since BIOS its not event starting. But do let me know.

What do you think?

ttzeng, 2019-01-25:
@LibertadCruz After you manual install the NUC by disabling the swupd auto update, your NUC should stay at 27230 after reboot. Then, add the `service-os` bundle, and copy the `/usr/lib/acrn/acrn.efi` to the ESP partition. Make sure you have also set the timeout value for the systemd.boot to pick up the SOS entry, otherwise the device will boot the native kernel by default, that might be the reason you see a blank screen, instead of a frozen screen.

coneypo, 2019-01-28:
### Issue Summary: 
NUC6CAY+ BIOS AYAPLCEL.86A.0059+ Clear Linux 27320 fail to boot ACRN

### Frequency: 
Always

### Preconditons:
NUC Model: NUC6CAY
BIOS Version: AYAPLCEL.86A.0059
https://downloadcenter.intel.com/downloads/eula/28478/BIOS-Update-AYAPLCEL-86A-?httpDown=https%3A%2F%2Fdownloadmirror.intel.com%2F28478%2Feng%2FAY0059.bio
Clear Linux: 27320

Steps:
	1. Update BIOS to AYAPLCEL.86A.0059
	2. Verify Clear Linux to 27320
	3. Follow  https://projectacrn.github.io/latest/getting-started/apl-nuc.html to set ACRN
	4. Reboot 
	5. Hang on the welcoming page that says "INTEL NUC"

### Expect result:
Boot to ACRN

### Actual result:
Hang in the "Intel NUC" screen

mwang106, 2019-01-28:
@LibertadCruz 
With 27230, you should have met this issue #2349

Mingyuan18, 2019-01-28:
[External_System_ID] ACRN-2512

gvancuts, 2019-03-05:
@LibertadCruz , could you let us know if this is still an issue?

Mingyuan18, 2019-03-22:
@LibertadCruz , May I ask whether this issue can be closed?

LibertadCruz, 2019-03-27:
Sorry I have been on vacations. Returning 8am on  March 27, I will check this and give an answer.

LibertadCruz, 2019-03-28:
The idea of this issue is that we cant get the newer version of CL and acrn-hypervisor to boot as it mentions on the getting started guide. This is still the case as of CL 28520.

gvancuts, 2019-03-28:
> The idea of this issue is that we cant get the newer version of CL and acrn-hypervisor to boot as it mentions on the getting started guide. This is still the case as of CL 28520. 
> 

Can you be more specific as to what the problem is?

Thanks,
Geoffroy

yifanddd, 2019-04-12:
[External_System_ID] ACRN-3376

fuzhongl, 2019-05-28:
@LibertadCruz 
Any more concern, please update the comments. If no more feedback within two weeks, this ticket will be closed by default. 
Appreciate for your understanding.

Mingyuan18, 2019-06-11:
No any feedback, Close this issue.

closed 2019-06-11 by Mingyuan18