projectacrn/acrn-hypervisor #2411: Cannot build master for NUC

rossburton, 2019-01-24:
For the nuc board and uefi firmware, the build fails:

```
| make[1]: Entering directory '/data/poky-tmp/master/work/intel_corei7_64-poky-linux/acrn-hypervisor/0.5-r0/git/hypervisor'
| Default configuration based on arch/x86/configs/nuc6cayh.config.
| Configuration written to /data/poky-tmp/master/work/intel_corei7_64-poky-linux/acrn-hypervisor/0.5-r0/build/hypervisor/.config.
| touch /data/poky-tmp/master/work/intel_corei7_64-poky-linux/acrn-hypervisor/0.5-r0/build/hypervisor/include/version.h
| ******* No ACPI info found *******
| Expected ACPI info header at bsp/include/nuc6cayh_acpi_info.h
|
| The ACPI info header for this board is not available. Please use
| an offline tool on the target board to generate a validated one.
| More details will be available soon.
```

miguelinux, 2019-01-24:
The solution is the same as #2414 
try with #2404

rossburton, 2019-01-25:
Erm yes, sorry!

closed 2019-01-25 by rossburton