projectacrn/acrn-hypervisor #2412: Top-level makefile cleans before building

rossburton, 2019-01-24:
It's impossible to do this:

```
make -C hypervisor menuconfig
make hypervisor
```

Because the first thing `make hypervisor` does is a clean, which removes the config file.

Mingyuan18, 2019-01-28:
[External_System_ID] ACRN-2606

junjiemao1, 2019-02-15:
@rossburton The top-level Makefile is specifically designed to be invoked by package builders to create, e.g. RPM, packages. If you would like to create a customized build, it would be better if you can invoke the component-specific Makefiles (like hypervisor/Makefile, devicemodel/Makefile, etc.) with command-line options.

Please kindly let me know if you have any requirement on the top-level Makefile. Thanks.

rossburton, 2019-02-15:
Speaking as a package builder, it's impossible to customise the .config before calling make hypervisor.

hwang37, 2019-02-19:
@jren1 @JasonChenCJ @dongyaozu to check this issue

hwang37, 2019-02-19:
Thanks Ross for the bug submission.
If the ask is about pre-defined configuration file, there're some ACRN configuration files defined under the folder below:
./hypervisor/arch/x86/configs/*

They are very simple, and rely the build process to generate the default configuration.
Can we just put the .config as the default configuration files? for your use?

fuzhongl, 2019-05-28:
@rossburton 
Any more concern, please update the comments. If no more feedback within two weeks, this ticket will be closed by default. 
Appreciate for your understanding.

rossburton, 2019-06-02:
My assertion that a 'build' target shouldn't enforce a clean still stands.  There should be standalone 'clean' and 'build' targets.

Mingyuan18, 2019-06-12:
Close with won't fix.

gvancuts, 2019-06-19:
@Mingyuan18 , I really wish you'd pay a little more attention to these kinds of bugs before closing them. I feel that @rossburton stance is very valid and more generally speaking, properly configuring, building and packaging ACRN is harder than it should be. Just closing the issue with no explanation at all is not how we should evaluate and discuss feedback we receive IMHO.

Mingyuan18, 2019-06-20:
@fuzhongl ,Please follow this issue. Thanks!

fuzhongl, 2019-11-20:
Will track with: https://github.com/projectacrn/acrn-hypervisor/issues/3779

mgcao, 2019-11-20:
OK, we'll consider how to decouple the "clean" with "build", it need add some dependency for Object and C/H files.

fuzhongl, 2020-04-27:
@rossburton Can this git issue be closed since patch was merged?
Thanks!

rossburton, 2020-04-27:
Doesn't look fixed to me:

https://github.com/projectacrn/acrn-hypervisor/blob/master/Makefile#L163

fuzhongl, 2020-05-05:
@junjiemao1 Could you please help to check this issue?
Thanks!

junjiemao1, 2020-05-05:
@binbinwu1 who is the owner of the top-level Makefile.

binbinwu1, 2020-05-06:
@jsun26intel  @terryzouhao Do you have any comments on this issue?

jsun26intel, 2020-05-06:
The first command is to initialize a .config file under acrn-hypervisor/hypervisor/ folder, the second command is to clean build folder under acrn-hypervisor/ folder. So we did not clean anything under acrn-hypervisor/hypervisor/ folder.

jsun26intel, 2020-05-06:
@rossburton

terryzouhao, 2020-05-06:
From functional perspective, current "make -C hypervisor menuconfig" + "make hypervisor" under acrn-hypervisor/ will not delete any config files, e.g., .config under acrn-hypervisor/hypervisor/build.

PS: only when you "make clean" under acrn-hypervisor/hypervisor, that will delete /build folder under /hypervisor.

But, i think the key idea/comment from @rossburton and @gvancuts maybe: we cannot mix clean in build/make, so could we move below clean action in makefile, and add dedicated step in doc/script, e.g., first step: "make clean",  then "make hypervisor" ? 
$(MAKE) -C $(T)/hypervisor HV_OBJDIR=$(HV_OUT) BOARD_FILE=$(BOARD_FILE) SCENARIO_FILE=$(SCENARIO_FILE) clean;

@mgcao do you think it is doable now, or still some concern for config file dependency ?

mgcao, 2020-05-07:
To me it is OK now, just for CI release need do some changes latter:
1. make clean
2. make

For developers, usually you don't need make clean.

So, now following can be removed, including HV and DM.
$(MAKE) -C $(T)/hypervisor HV_OBJDIR=$(HV_OUT) BOARD_FILE=$(BOARD_FILE) SCENARIO_FILE=$(SCENARIO_FILE) clean

mgcao, 2020-05-07:
I'll submit a patch for that.

terryzouhao, 2020-05-07:
+@wenlingz and @NanlinXie to see if any concern for release.

wenlingz, 2020-05-07:
LGTM

> +@wenlingz and @NanlinXie to see if any concern for release.

It looks no influence for CI

terryzouhao, 2020-05-20:
@rossburton @gvancuts , could you double check the [PR#4826](https://github.com/projectacrn/acrn-hypervisor/pull/4826) from @mgcao , if it meet the initiative of de-couple Clean and Make, welcome comment and ideas, thanks.

closed 2020-06-28 by fuzhongl