projectacrn/acrn-hypervisor #2455: host call stack disappear when dumping

totoroow, 2019-01-30:
When an exception happens in ACRN, host call stack should be dumped to help debug, but it disappears now.

For example, it I type `reboot` in ACRN shell:

```
 21 ^MACRN:\>reboot^M
 20 ^Mtrigger crash, divide by 0 ...^M
 19 ^M
 18 ^M
 17 ^M================================================================================
 16 ^M=
 15 ^M= Unhandled exception: 0 (Divide Error)
 14 ^M^M
 13 ^MHost Registers:^M
 12 ^M=  Vector=0x0000000000000000  RIP=0x0000000000240A85
 11 ^M=     RAX=0x0000000000000001  RBX=0x0000000000000000  RCX=0x0000000000000000
 10 ^M=     RDX=0x00000000000003F8  RDI=0x000000000A5702B0  RSI=0x00000000000003F8
  9 ^M=     RSP=0x000000000A411610  RBP=0x000000000A4117B8  RBX=0x0000000000000000
  8 ^M=      R8=0x000000000A4117E0   R9=0x0000000000000000  R10=0x0000000000000000
  7 ^M=     R11=0x0000000000000000  R12=0x0000000000000000  R13=0x0000000000000000
  6 ^M=  RFLAGS=0x0000000000010206  R14=0x0000000000000000  R15=0x0000000000000000
  5 ^M= ERRCODE=0x0000000000000000   CS=0x0000000000000008   SS=0x000000000A4116A8
  4 ^M^M
  3 ^M================================================================================
  2 ^M^M
  1 ^MHost Stack: CPU_ID = 0^M
  0 ^Maddr(0xa411610)   0x000000000a4117b8  0x00000000000003f8  0x000000000a5702b0  0x000000000a4117e0^M
  1 ^Maddr(0xa411630)   0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000^M
  2 ^Maddr(0xa411650)   0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000^M
  3 ^Maddr(0xa411670)   0x0000000000000000  0x0000000000240a85  0x0000000000000008  0x0000000000010206^M
  4 ^Maddr(0xa411690)   0x000000000a4116a8  0x0000000000000010  0x0000000000240a7b  0x000000000a4117f0^M
  5 ^Maddr(0xa4116b0)   0x0000000100000000  0x2072656767697274  0x64202c6873617263  0x7962206564697669^M
  6 ^Maddr(0xa4116d0)   0x0a0d2e2e2e203020  0x0000000000000000  0x0000000000000000  0x0000000000000000^M
  7 ^Maddr(0xa4116f0)   0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000^M
  8 ^Maddr(0xa411710)   0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000^M
  9 ^Maddr(0xa411730)   0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000^M
 10 ^Maddr(0xa411750)   0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000^M
 11 ^Maddr(0xa411770)   0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000^M
 12 ^Maddr(0xa411790)   0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000^M
 13 ^Maddr(0xa4117b0)   0x0000000000000000  0x000000000a411ea8  0x000000000023eaef  0x0000000000000000^M
 14 ^Maddr(0xa4117d0)   0x000000000a56e105  0x0000000000000000  0x000000000a4117f0  0x0000000100000000^M
 15 ^Maddr(0xa4117f0)   0x000000000a411e33  0x0000000000000000  0x0000000000000000  0x0000000000000000^M
 16 ^M^M
 17 ^MHost Call Trace:^M

 ^M================================================================================
  8 ^M
  7 ^MGuest Registers:^M
  6 ^M= VM ID 0 ==== vCPU ID 0 ===  pCPU ID 0 ====world 0 =============^M
  5 ^M= RIP=0x000000004c827cef  RSP=0x000000004d7a6110 RFLAGS=0x0000000000000002^M
  4 ^M= CR0=0x0000000080010013  CR2=0x0000000000000000  CR3=0x000000004d001000^M
  3 ^M= RAX=0x00000000fee00900  RBX=0x0000000000001600  RCX=0x000000000000001b^M
  2 ^M= RDX=0x0000000000000000  RDI=0x000000004ca50018  RSI=0x0000000000000001^M
  1 ^M= RBP=0x000000004d7a61c0  R8=0x0000000000000000  R9=0x0000000000000000^M
  0 ^M= R10=0x0000000030726670  R11=0x000000004d7a65e0  R12=0x000000004ca59098^M
  1 ^M= R13=0x0000000000000002  R14=0x000000004d7a6258  R15=0x0000000000000040^M
  2 ^M^M
  3 ^M^M
  4 ^MGuest Stack:^M
  5 ^MDump stack for vcpu 0, from gva 0x000000004d7a6110^M
  6 ^Mguest_rsp(0x4d7a6110):  0x00000000000806c0  0x000000004d7e5e00  0x000000004b0e9e98  0x0000000000000004^M
  7 ^Mguest_rsp(0x4d7a6130):  0x000000004d7e5e60  0x000000004c827d50  0x0000000000000010  0x0000000000000000^M
  8 ^Mguest_rsp(0x4d7a6150):  0x000000004b0e9e98  0x000000004c82759e  0x000000004ca50018  0x000000004c827484^M
  9 ^Mguest_rsp(0x4d7a6170):  0x000000001bbdede8  0x000000004d7a61c0  0x0000000000000001  0x000000004ca2659e^M
 10 ^Mguest_rsp(0x4d7a6190):  0x0000000000001600  0x000000004ca2659e  0x000000004d7a6290  0x000000004acf8b98^M
 11 ^Mguest_rsp(0x4d7a61b0):  0x800000000000000e  0x000000004ad1b878  0x800000000000000e  0x000000004ad1b878^M
 12 ^Mguest_rsp(0x4d7a61d0):  0x000000004ad1ca18  0x800000000000000e  0x000000004d7a6260  0x00000000013cb35b^M
 13 ^Mguest_rsp(0x4d7a61f0):  0x000000004d7a65c0  0x000000004ca261c5  0x0000000000000001  0x000000004d7a6258^M
 14 ^Mguest_rsp(0x4d7a6210):  0x000000004e4af3a0  0x000000004ca217a0  0x0000000000000004  0x000000004ca217a0^M
 15 ^Mguest_rsp(0x4d7a6230):  0x000000004acf8000  0x000000000002200f  0x0000000000000004  0x0000000000000000^M
 16 ^Mguest_rsp(0x4d7a6250):  0x000000000000037f  0x0000000000000000  0x000000000000037f  0x0000000000000000^M
 17 ^Mguest_rsp(0x4d7a6270):  0x0000000000000000  0x0000ffff00001f80  0x0000000000000000  0x000000000000ffff^M
 18 ^Mguest_rsp(0x4d7a6290):  0x0000000000000000  0x0000000000000000  0x0000000000800000  0x000000000000ffff^M
 19 ^Mguest_rsp(0x4d7a62b0):  0x0000000000000000  0x000000000000ffff  0x00000000ff780000  0x000000000000ffff^M
 20 ^Mguest_rsp(0x4d7a62d0):  0x0000000000000000  0x000000000000ffff  0x0000000000005118  0x000000000000ffff^M
 21 ^Mguest_rsp(0x4d7a62f0):  0x00000000ffea0482  0x000000000000ffff  0x11d293ca8be4df61  0x8c2b0398e0000daa^M
 22 ^M^M
 23 ^MGuest Call Trace: **************************************^M

```

totoroow, 2019-01-30:
[External_System_ID] ACRN-2627

Mingyuan18, 2019-02-27:
The bug cannot reproduced with this build:
Installed version: 28010

closed 2019-02-27 by Mingyuan18