projectacrn/acrn-hypervisor #2457: Compilation on Fedora 29 (for UEFI platforms) is broken

gvancuts, 2019-01-30:
Attempting to compile the ACRN hypervisor for UEFI platforms on Fedora 29 will fail as follows:
```
<snip>
echo "building hypervisor as EFI executable..."
building hypervisor as EFI executable...
make -C /home/gvancuts/work/acrn/acrn-hypervisor/efi-stub HV_OBJDIR=/home/gvancuts/work/acrn/acrn-hypervisor/build/hypervisor EFI_OBJDIR=/home/gvancuts/work/acrn/acrn-hypervisor/build/efi-stub
make[1]: Entering directory '/home/gvancuts/work/acrn/acrn-hypervisor/efi-stub'
find: ‘/usr/libexec/initscripts/legacy-actions/auditd’: Permission denied
dirname: missing operand
Try 'dirname --help' for more information.
[ ! -e /home/gvancuts/work/acrn/acrn-hypervisor/build/efi-stub/boot.o ] && mkdir -p /home/gvancuts/work/acrn/acrn-hypervisor/build/efi-stub/; \
cc  -I/home/gvancuts/work/acrn/acrn-hypervisor/build/hypervisor/include -I. -c -I. -I.. -I../hypervisor/include/arch/x86/guest -I/usr/include/efi -I/usr/include/efi/x86_64 -I../hypervisor/include/public -I../hypervisor/include/lib -I../hypervisor/bsp/include/uefi -DEFI_FUNCTION_WRAPPER -fPIC -fshort-wchar -ffreestanding -Wall -I../fs/ -Dx86_64 -O2 -include config.h -mno-mmx -mno-sse -mno-sse2 -mno-80387 -mno-fp-ret-in-387 -mno-red-zone  boot.c -o /home/gvancuts/work/acrn/acrn-hypervisor/build/efi-stub/boot.o
In file included from boot.c:34:
/usr/include/efi/efi.h:41:10: fatal error: efibind.h: No such file or directory
 #include "efibind.h"
          ^~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:114: /home/gvancuts/work/acrn/acrn-hypervisor/build/efi-stub/boot.o] Error 1
make[1]: Leaving directory '/home/gvancuts/work/acrn/acrn-hypervisor/efi-stub'
make: *** [Makefile:52: hypervisor] Error 2
```

The root cause is that the `gnu-efi` packages now put architecture-dependent files using `x64` and not `x86_64`.

The following patch (**not** suitable for upstreaming) works around that problem on Fedora 29, for `x86_64` platforms):
```
diff --git a/efi-stub/Makefile b/efi-stub/Makefile
index 5b87d49b..dfc64843 100644
--- a/efi-stub/Makefile
+++ b/efi-stub/Makefile
@@ -52,14 +52,14 @@ endif
 # its tools and libraries in different folders. The next couple of
 # variables will determine and set the right path for both the
 # tools $(GNUEFI_DIR) and libraries $(LIBDIR)
-GNUEFI_DIR := $(shell find $(SYSROOT)/usr/lib* -name elf_$(ARCH)_efi.lds -type f | xargs dirname)
+GNUEFI_DIR := $(shell find $(SYSROOT)/usr/lib* -name elf_x64_efi.lds -type f | xargs dirname)
 LIBDIR := $(subst gnuefi,,$(GNUEFI_DIR))
-CRT0 := $(GNUEFI_DIR)/crt0-efi-$(ARCH).o
-LDSCRIPT := $(GNUEFI_DIR)/elf_$(ARCH)_efi.lds
+CRT0 := $(GNUEFI_DIR)/crt0-efi-x64.o
+LDSCRIPT := $(GNUEFI_DIR)/elf_x64_efi.lds
 
 INCDIR := $(SYSROOT)/usr/include
 
-CFLAGS=-I. -I.. -I../hypervisor/include/arch/x86/guest -I$(INCDIR)/efi -I$(INCDIR)/efi/$(ARCH) \
+CFLAGS=-I. -I.. -I../hypervisor/include/arch/x86/guest -I$(INCDIR)/efi -I$(INCDIR)/efi/x64 \
                -I../hypervisor/include/public -I../hypervisor/include/lib -I../hypervisor/bsp/include/uefi \
                -DEFI_FUNCTION_WRAPPER -fPIC -fshort-wchar -ffreestanding \
                -Wall -I../fs/ -D$(ARCH) -O2 \
```

Mingyuan18, 2019-02-14:
[External_System_ID] ACRN-2680

fuzhongl, 2019-04-25:
Hi gvancuts
Could you please send out the guide about the fix? 
Then, we will update GSG accordingly.
Thanks!

gvancuts, 2019-04-25:
> Could you please send out the guide about the fix?

The workaround I use is to patch the `efi-stub/Makefile` using the patch below (see [1]). Unfortunately, this is not suitable to be merged in the repo as it would break other distros. Our best option would be to add a note in the GSG explaining this, is that what you are proposing to do?

Thanks!
Geoffroy

[1]
```
index 5b87d49b..dfc64843 100644
--- a/efi-stub/Makefile
+++ b/efi-stub/Makefile
@@ -52,14 +52,14 @@ endif
 # its tools and libraries in different folders. The next couple of
 # variables will determine and set the right path for both the
 # tools $(GNUEFI_DIR) and libraries $(LIBDIR)
-GNUEFI_DIR := $(shell find $(SYSROOT)/usr/lib* -name elf_$(ARCH)_efi.lds -type f | xargs dirname)
+GNUEFI_DIR := $(shell find $(SYSROOT)/usr/lib* -name elf_x64_efi.lds -type f | xargs dirname)
 LIBDIR := $(subst gnuefi,,$(GNUEFI_DIR))
-CRT0 := $(GNUEFI_DIR)/crt0-efi-$(ARCH).o
-LDSCRIPT := $(GNUEFI_DIR)/elf_$(ARCH)_efi.lds
+CRT0 := $(GNUEFI_DIR)/crt0-efi-x64.o
+LDSCRIPT := $(GNUEFI_DIR)/elf_x64_efi.lds
 
 INCDIR := $(SYSROOT)/usr/include
 
-CFLAGS=-I. -I.. -I../hypervisor/include/arch/x86/guest -I$(INCDIR)/efi -I$(INCDIR)/efi/$(ARCH) \
+CFLAGS=-I. -I.. -I../hypervisor/include/arch/x86/guest -I$(INCDIR)/efi -I$(INCDIR)/efi/x64 \
                -I../hypervisor/include/public -I../hypervisor/include/lib -I../hypervisor/bsp/include/uefi \
                -DEFI_FUNCTION_WRAPPER -fPIC -fshort-wchar -ffreestanding \
                -Wall -I../fs/ -D$(ARCH) -O2 \
```

fuzhongl, 2019-04-26:
Thank you, Geoffroy!
Will add it in FAQ.

BR.
Fuzhong

Mingyuan18, 2019-05-05:
Add in FAQ

closed 2019-05-05 by Mingyuan18