projectacrn/acrn-hypervisor #2490: systemd virtualisation detection doesn't work

rossburton, 2019-02-01:
systemd has a `systemd-detect-virt` command that determines what virtualisation or container system is in place.  I don't expect it to say 'acrn' just yet because it doesn't know the magic cpuid string yet (I've a patch for that), but it should say 'other'.  But apparently not, from both the SOS and UOS it always returns 'none'.

https://github.com/systemd/systemd/blob/master/src/basic/virt.c#L24 is the logic.  Not sure whether systemd or acrn is at fault here.

gvancuts, 2019-02-01:
@miguelinux , this is a problem we had identified a while ago. Were you able to look into that?

There is something odd about `systemd-detect-virt` in general. If I run Clear Linux in a container, it also returns `none`. To test this: `sudo docker run clearlinux bash -c "systemd-detect-virt"`

As a quick test, I tried to use the `systemd-detect-virt` binary from Fedora 29 in that container, it also returned `none`.

gvancuts, 2019-02-13:
Hi @miguelinux , can you let us know whether you're looking at this?

miguelinux, 2019-02-27:
This is a feature related to ACRN, not to CL or systemd.

The hypervisor should return that logic when the cpuid is asked.

https://github.com/projectacrn/acrn-hypervisor/blob/master/hypervisor/arch/x86/guest/vcpuid.c

Hypervisor Present Bit:
        Bit 31 of ECX of CPUID leaf 0x1.

@dongyaozu @anthonyzxu 

Any suggestion?

https://lists.projectacrn.org/g/acrn-dev/message/7070
https://lists.projectacrn.org/g/acrn-dev/message/5565

miguelinux, 2019-02-27:
Is `cpuid -1 | grep 'hypervisor guest status'` or `cpuid -1 -r | grep '^   0x00000001.*ecx=0x[89a-f]'` working on ACRN?

https://unix.stackexchange.com/questions/349968/how-to-check-if-the-hypervisor-present-bit-is-set-using-cpuid

rossburton, 2019-02-27:
As the SOS:
```
root@bob:~# cpuid -l 0x40000000
CPU 0:
   hypervisor_id = "ACRNACRNACRN"
CPU 1:
   hypervisor_id = "ACRNACRNACRN"
CPU 2:
   hypervisor_id = "ACRNACRNACRN"
CPU 3:
   hypervisor_id = "ACRNACRNACRN"
CPU 4:
   hypervisor_id = "ACRNACRNACRN"
CPU 5:
   hypervisor_id = "ACRNACRNACRN"
CPU 6:
   hypervisor_id = "ACRNACRNACRN"
CPU 7:
   hypervisor_id = "ACRNACRNACRN"

root@bob:~# cpuid -1 | grep 'hypervisor guest status'
      hypervisor guest status                 = false

root@bob:~# cpuid -1 -r | grep 0x00000001
   0x00000001 0x00: eax=0x000506e3 ebx=00100800 ecx=0x7ff8738b edx=0xbfcbfb7f
```

As a Clear UOS:

```
# cpuid -l 0x40000000
CPU 0:
   hypervisor_id = "ACRNACRNACRN"
CPU 1:
   hypervisor_id = "ACRNACRNACRN"
CPU 2:
   hypervisor_id = "ACRNACRNACRN"
CPU 3:
   hypervisor_id = "ACRNACRNACRN"
CPU 4:
   hypervisor_id = "ACRNACRNACRN"
CPU 5:
   hypervisor_id = "ACRNACRNACRN"
CPU 6:
   hypervisor_id = "ACRNACRNACRN"
CPU 7:
   hypervisor_id = "ACRNACRNACRN"

# cpuid -1 | grep 'hypervisor guest status'
      hypervisor guest status                 = false

# cpuid -1 -r | grep '^ +0x00000001'
   0x00000001 0x00: eax=0x000506e3 ebx=0x00100800 ecx=0x7ff8738b edx=0xbfcbfb7f
```

(they're identical)

miguelinux, 2019-02-27:
> As the SOS:
> root@bob:~# cpuid -1 | grep 'hypervisor guest status'
>       hypervisor guest status                 = false

Thanks,

Then ACRN needs to fix this issue, the hypervisor should return true
here.

anthonyzxu, 2019-02-27:
We knew this issue,  we will fix it soon

xinyunliu, 2019-02-28:
@rossburton @anthonyzxu, I have a fix in hand, will submit the patch for review. Thanks for open the issue.

xinyunliu, 2019-02-28:
with the patch, in clearlinux 27310:
``` 
$ systemd-detect-virt
vm-other
````
with the tool here: https://github.com/xyliu/c_exp/tree/master/cpuid
```
$ ~/cpuid
CPU vendor = GenuineIntel
in hypervisor: Y
raw: (EAX,EBX,ECX,EDX)=(000806e9,03100800,fff8738b,bfcbfb7f)
hypervisor = ACRNACRNACRN
40000000 hypervisor = ACRNACRNACRN
d0000100 hypervisor = �

```

Mingyuan18, 2019-02-28:
[External_System_ID] ACRN-2834

rossburton, 2019-02-28:
That's great, I've already got a patch for systemd so I'll test that today.

Ross

rossburton, 2019-02-28:
Miguel: you seem to understand this well.  Is there meant to be a way to differentiate between the SOS and UOS using the cpuid data?

rossburton, 2019-02-28:
With https://github.com/systemd/systemd/pull/11857:

```
$ systemd-detect-virt
acrn
```

miguelinux, 2019-02-28:
> Miguel: you seem to understand this well.  Is there meant to be a way to differentiate between the SOS and UOS using the cpuid data?
> 

Well, not really as you are under a Hypervisor, it is detected that is
ACRN, and the OS is SOS or UOS.

So cpuid only thell you if you are under an hypervisor.

The OS type SOS or UOS, could be detected by acrnctl or acrn-dm I guess.

SOS is Dom0, and UOS is DomX, X > 0

gvancuts, 2019-03-01:
> With [systemd/systemd#11857](https://github.com/systemd/systemd/pull/11857):
> 
> ```
> $ systemd-detect-virt
> acrn
> ```

Nice, thanks! (just need this to land in Clear Linux now ;-))

miguelinux, 2019-03-01:
> > With [systemd/systemd#11857](https://github.com/systemd/systemd/pull/11857):
> > 
> > ```
> > $ systemd-detect-virt
> > acrn
> > ```
> 
> Nice, thanks! (just need this to land in Clear Linux now ;-))
> 

Next week when new systemd landed ;-)

gvancuts, 2019-03-01:
> Miguel: you seem to understand this well. Is there meant to be a way to differentiate between the SOS and UOS using the cpuid data?
> Well, not really as you are under a Hypervisor, it is detected [...]

It probably does not make sense to have `acrnctl` or `acrn-dm` installed in a UOS.

I can think of a couple of other ways we could detect if we're in the SOS or UOS:
1. Check if we see the BSP (bootstrap processor), I believe it can only be seen from within the SOS* (the best way I found to check its presence is to use `rdmsr --all 0x1B --bitfield 8:8`, if there is a `1` in the list, then we see the BSP.
2. Check the presence of `/dev/acrn_hvm` or `/sys/class/vhm/acrn_vhm`. If it's there, you're in the SOS **

Out of curiosity, why do you want to know that?

*: I'm not sure if having a pre-defined, static VM that is automatically started by the hypervisor could change that.
**: to be verified whether a **Privileged VM** would have these or not.

miguelinux, 2019-03-09:
Added in https://github.com/clearlinux-pkgs/systemd/commit/be730e85102e9f50b888fd2ab06bf51ae068ef81

miguelinux, 2019-03-09:
the changes are since 28230

rossburton, 2019-03-20:
Is there a reason why that patch hasn't been merged to master yet?

rossburton, 2019-03-20:
Merged into acrn-hypervisor master as of 0943a836bd7bcb57b22e79d3cc65b1d25940a4e1.

closed 2019-03-20 by rossburton