projectacrn/acrn-hypervisor #2507: iTCO_WDT driver fails under acrn

rossburton, 2019-02-05:
My Skull Canyon has the iTCO watchdog, and if that kernel driver is enabled (`CONFIG_ITCO_WDT`) then the kernel hangs on boot:

```
iCTO_wdt: MSR access error: WRMSR to 0x80b tried to write 0x0000 at rIP 0xfffffffac046f5d (native_apic_msr_eoi_write)
```

Followed by a looping oops and eventual reboot.  The same kernel works outside of the acrn hypervisor, and disabling the driver lets the kernel boot.

rossburton, 2019-02-05:
A friendly kernel engineers says

> That part sounds like a dodgy hypervisor, not allowing writes to the on-CPU local interrupt controller...

rossburton, 2019-02-05:
CC @moto-timo

Mingyuan18, 2019-02-14:
[External_System_ID] ACRN-2679

Mingyuan18, 2019-05-23:
@rossburton , This issue has been fixed, please confirm it. Thanks!

Mingyuan18, 2019-06-05:
@rossburton , Please try this issue in below build, It was fixed. If there is no reply in two weeks, we will close this issue. Thanks!
Clean Linux: 28920
ACRN tag: acrn-2019w17.2-160000p

rossburton, 2019-06-05:
Sorry but disabling the driver to work around a bug in the hypervisor isn't a solution the fact that there's a bug in the hypervisor.

Luckily the hypervisor appears to have been fixed in the meantime. My SKL with iTCO_wtd driver enabled is booting successfully.

closed 2019-06-05 by rossburton