projectacrn/acrn-hypervisor #2515: Display corruption in SOS on KBL

rossburton, 2019-02-12:
On a KBL NUC with current Clear SOS and the following kernel options:

```
i915.avail_planes_per_pipe=0x00000F
i915.domain_plane_owners=0x011111110000
i915.enable_gvt=1
i915.nuclear_pageflip=1
```

There is display corruption in the SOS under GNOME.  Specifically, there are multiple pointers visible that often flicker in and out of visibility based on other changes to the display, such as a cursor flashing.

Of note is that `/sys/kernel/debug/drm/0/i915_display_info` reports that the cursor plane isn't available.  Attached are the `state` and `i915_display_info` files for both native Clear and Clear as SOS showing the differences.  As I've a single output connected so a single pipe is being used, and the GVT-g options were allowing the use of the full pipe, this isn't expected.

[display-info-clear.txt](https://github.com/projectacrn/acrn-hypervisor/files/2858121/display-info-clear.txt)
[display-info-sos.txt](https://github.com/projectacrn/acrn-hypervisor/files/2858122/display-info-sos.txt)
[state-clear.txt](https://github.com/projectacrn/acrn-hypervisor/files/2858123/state-clear.txt)
[state-sos.txt](https://github.com/projectacrn/acrn-hypervisor/files/2858124/state-sos.txt)

minhe1, 2019-02-14:
@rossburton , the cursor plane is actually one of universal planes on the pipe. When enabling the plane restriction feature, we need to disable the cursor plane, so that the planes can be share to other OS.

Mingyuan18, 2019-03-21:
@rossburton, May I ask whether this issue can be closed？

Mingyuan18, 2019-03-25:
[External_System_ID] ACRN-3128

fuzhongl, 2019-05-28:
@rossburton 
Any more concern, please update the comments. If no more feedback within two weeks, this ticket will be closed by default. 
Appreciate for your understanding.

rossburton, 2019-06-04:
Just tested, works correctly with latest kernel/acrn.

closed 2019-06-04 by rossburton

kriskep, 2019-12-23:
I am still observing display corruption under GNOME with the latest acrn/kernel.
acrn version: acrn-2019w50.5
kernel: 4.19.78-98.iot-lts2018-sos

fuzhongl, 2019-12-24:
@kriskep Which CL build are you using? You use  CL E2E kernel; not acrn-kernel. Correct?
Is the issue still there if not launch Guest VM?
Thanks!

gvancuts, 2019-12-24:
> @kriskep Which CL build are you using? You use CL E2E kernel; not acrn-kernel. Correct?
> Is the issue still there if not launch Guest VM?

FWIW, I see this too using the CL E2E kernel and even before launching a User VM.

fuzhongl, 2019-12-25:
> > @kriskep Which CL build are you using? You use CL E2E kernel; not acrn-kernel. Correct?
> > Is the issue still there if not launch Guest VM?
> 
> FWIW, I see this too using the CL E2E kernel and even before launching a User VM.

@gvancuts  Thanks for reply!
As you know, there are still some acrn related kernel patches not merged into E2E kernel, Could you have a try with acrn-kernel:
[](https://github.com/projectacrn/acrn-kernel)
Thanks!

gvancuts, 2019-12-26:
> As you know, there are still some acrn related kernel patches not merged into E2E kernel
Do you expect that to be closed at some point?

> Could you please have a try with [acrn-kernel](https://github.com/projectacrn/acrn-kernel) ?
I'll let @kriskep confirm it works correctly for him with the ACRN kernel (I'll also report out when I get to it)

kriskep, 2020-01-06:
This was still seen with acrn-kernel (sos) 4.19.73-PKT-190919T160721Z.
Launching a guest VM doesn't seem to make a difference.