projectacrn/acrn-hypervisor #2518: Service OS kernel parameters for i915 need a clean-up (EFI platforms)

gvancuts, 2019-02-13:
The i915 driver parameters used in `acrn.conf` (for EFI platforms) need a clean-up:
- Some are inconsistent
- A few are obsolete

rossburton, 2019-02-13:
That was spooky, I literally just spent a few minutes doing an identical patchset.

Well, almost identical.  'i915.enable_guc' doesn't exist either:

```
$ grep enable_guc drivers/gpu/drm/i915/i915_params.h
	param(int, enable_guc_loading, 0) \
	param(int, enable_guc_submission, 0) \
```

ttzeng, 2019-02-13:
@gvancuts glad you kick off the clean up process... :+1: 
@rossburton I suspect we can check the presence of i915 parameters using grep, as some may be disabled while building the driver.  For example, I'm sure `enable_guc` is there, you could check the available parameters through `/sys/modules/i915/parameters` interface.

rossburton, 2019-02-13:
Yes, my fault, I was looking at the wrong branch!

Mingyuan18, 2019-02-14:
[External_System_ID] ACRN-2676

gvancuts, 2019-02-14:
**Update**: although the flags pertaining to the plane restriction feature look inconsistent, there is rationale behind them which is to ensure that Weston initializes the CRTC all the pipes, something that would *not* happen if the Service OS does not have at least one plane on each pipe. That plane assignment is actually enforced by the i915 driver so setting `i915.avail_planes_per_pipe` to `0x00000F` for example will automatically be converted to `0x01010F`, see this patch for more details: https://lists.projectacrn.org/g/acrn-dev/message/10800

rossburton, 2019-02-14:
Note that the https://projectacrn.github.io/latest/tutorials/agl-vms.html has more references that need cleaning.

gvancuts, 2019-02-15:
> Note that the https://projectacrn.github.io/latest/tutorials/agl-vms.html has more references that need cleaning.

Thanks @rossburton , I have updated that tutorial (and two others as well) and updated my PR.

fuzhongl, 2019-05-14:
> > Note that the https://projectacrn.github.io/latest/tutorials/agl-vms.html has more references that need cleaning.
> 
> Thanks @rossburton , I have updated that tutorial (and two others as well) and updated my PR.

Hi Geoffroy
Can this issue be closed?
Thanks!

gvancuts, 2019-05-14:
I believe so, closing

closed 2019-05-14 by gvancuts