projectacrn/acrn-hypervisor #2526: Hypervisor crash when booting UOS with acrnlog running with mem loglevel=6

Mingyuan18, 2019-02-14:
If we use loglevel 3 6 to change the mem loglevel to 6, we could hit page fault in HV.

Mingyuan18, 2019-02-14:
[External_System_ID] ACRN-2678

gvancuts, 2019-03-05:
I have come across a different scenario that triggers a page fault: `loglevel set 6`. This is not 100% reproducible but it occurs frequently enough. When that happens, I typically have a VM running and it is frozen at that point.

I do realize that `loglevel set 6` is not a valid command :) ... but we should throw an error on the shell rather than generate a page fault that can freeze a UOS. I have also seen this freeze completely my Service OS at one point.

Is this covered by this bug report or should I file a new one?

fyin1, 2019-03-06:
command "loglevel set 6" does set the serial console log level to 6. Which could make lots of message printed on serial console. Printing message to serial console takes time. So it could impact the timing of the hypervisor that could make hypervisor can't work correctly. This should be expected behavior.
But "loglevel set 6" is not invalid command. It could help when debugging boot time issue (which means we don't care whether the system could boot. We just want to know what's going on when system is booting). But we shouldn't abuse it.

So we have command "loglevel 3 6" which keep serial console loglevel to 3 and raise memory logger loglevel to 6. Which should work at any situation. This patch fix the issue for command "loglevel 3 6".

gvancuts, 2019-03-06:
> command "loglevel set 6" does set the serial console log level to 6.

That is incorrect. Setting the serial console loglevel to 6 is done via `loglevel 6`. Calling `loglevel set 6` actually sets the console loglevel to `0` (which must be an artefact of the conversion from string to decimal done by `strtol_deci` (see https://github.com/projectacrn/acrn-hypervisor/blob/master/hypervisor/debug/shell.c#L1193-L1218)

Here is the output on my system:
```
ACRN:\>loglevel
console_loglevel: 3, mem_loglevel: 5, npk_loglevel: 5
ACRN:\>loglevel set 6
ACRN:\>loglevel
console_loglevel: 0, mem_loglevel: 6, npk_loglevel: 5
ACRN:\>loglevel 6
[39257025us][cpu=3][sev=6][seq=106412]:Exit Reason: 0x0000000000000001 
[39262025us][cpu=1][sev=6][seq=106413]:Exit Reason: 0x0000000000000001 
[39264240us][cpu=3][sev=6][seq=106414]:Exit Reason: 0x0000000000000020 
[39271444us][cpu=1][sev=6][seq=106415]:vlapic: lapic MSI addr: fee02000 msg: 4022
[39278695us][cpu=3][sev=6][seq=106416]:vlapic: cpu[3] wrmsr: 6e0 wval=2c11c7aa
[39286923us][cpu=1][sev=6][seq=106417]:vlapic: lapic MSI physical dest 2, vec 34
[39293658us][cpu=0][sev=6][seq=106418]:Exit Reason: 0x0000000000000001 
[39294876us][cpu=3][sev=6][seq=106419]:Exit Reason: 0x0000000000000020 ACRN:\>
[39302998us][cpu=1][sev=6][seq=106420]:dev-assign: irq=0x79 MSI VR: 0x22-0x37
[39318185us][cpu=3][sev=6][seq=106421]:vlapic: cpu[3] wrmsr: 6e0 wval=3453e30
[39326026us][cpu=1][sev=6][seq=106422]: vmsi_addr: 0xfee02000 vmsi_data: 0x4022
<snip>
```
 
> Which could make lots of message printed on serial console.

Yes, it does :-)

> Printing message to serial console takes time. So it could impact the timing of the hypervisor that could make hypervisor can't work correctly. This should be expected behavior.

I don't think it's an appropriate behaviour for the hypervisor to crash if there are timing issues when printing too much log output, it would be more appropriate to drop some (and ideally post a message that the logs are incomplete when that happens).

> But "loglevel set 6" is not invalid command.

It is, see above. I should have used `loglevel 6` instead to get what I wanted (but the shell help confused me a bit).

gvancuts, 2019-03-06:
> This patch fix the issue for command "loglevel 3 6".

Now that I've wrapped my head around the original issue and what I reported, I realize this is the exact problem I hit. Your patch should therefore close my issue too! :-)

Mingyuan18, 2019-05-13:
The bug cannot reproduced with below build:
root@sos-kbl-nuc-nvme ~ # swupd info
SOS version: 29350
acrn tag: 2019w19.4.140000p
Kernel:  4.19.40-49.iot-lts2018-sos

closed 2019-05-13 by Mingyuan18