projectacrn/acrn-hypervisor #2527: [KBLNUC][HV]System will crash when run crashme.

mwang106, 2019-02-14:
1.Environment
[Board]: KBL NUC

Installed version: 27600
Version URL: http://linux-ftp.sh.intel.com/pub/mirrors/clearlinux/update
Content URL: http://linux-ftp.sh.intel.com/pub/mirrors/clearlinux/update

2.Reproduce step:
--------------------------------------------------------------
1). Boot up SOS/UOS.

2). Run ./crashme +2000 666 50 1:0:0

3. Current result:
--------------------------------------------------------------
System will crash when run crashme.

4. Expected result:
--------------------------------------------------------------
System works well,no crash,no hang.

5. repetition rate
--------------------------------------------------------------
80%

6.Logs:

mwang106, 2019-02-14:
[External_System_ID] ACRN-2669

Mingyuan18, 2019-05-15:
The bug cannot reproduced with E2E build info:
SOS version: 29350
acrn tag: 2019w19.4.140000p
Linux sos-kbl-nuc-nvme 4.19.40-49.iot-lts2018-sos #1023

closed 2019-05-15 by Mingyuan18