projectacrn/acrn-hypervisor #2542: GVT-g not fully functional on SKL

rossburton, 2019-02-15:
Sky Lake is Gen9, Kaby Lake is Gen9.5.

So in theory, it shouldn't be *that* difficult to enable SKL.  However a lot of the kernel code is guarded with IS_KABYLAKE() so doesn't even attempt to work on SKL.

Mingyuan18, 2019-03-21:
[External_System_ID] ACRN-3083

fuzhongl, 2019-05-28:
@rossburton Sky Lake isn't the supported HW.
Any more concern, please update the comments. If no more feedback within two weeks, this ticket will be closed by default. 
Appreciate for your understanding.

rossburton, 2019-06-04:
For what it's worth the only piece of SKL enabling unmerged is the top two commits on this branch:

https://github.com/projectacrn/acrn-kernel/compare/acrn_skl_test

I'd love to see them merged, even if SKL isn't a supported platform these patches make it work.

fuzhongl, 2019-06-04:
@rossburton  Let me check it internally and back to you.
Thanks!

fuzhongl, 2020-06-28:
@rossburton  Please try with release_2.0
Thanks!

rossburton, 2020-06-29:
Pinging @moto-timo as I don't work for Intel anymore.

closed 2021-03-15 by fuzhongl