projectacrn/acrn-hypervisor #2548: [APLNUC/KBLNUC][GVT][SOS/LAAG] Weston fails to play video in SOS and UOS

mwang106, 2019-02-18:
Environment:
------------

HW: APLNUC

Build: https://download.clearlinux.org/releases/27840/clear/

 

SOS version: 
root@sos-dom0 ~/test # swupd info
Installed version: 27840
Version URL: https://cdn.download.clearlinux.org/update/
Content URL: https://cdn.download.clearlinux.org/update/
root@sos-dom0 ~/test # uname -a
Linux sos-dom0 4.19.20-14.iot-lts2018-sos #1 SMP PREEMPT Fri Feb 15 05:33:46 UTC 2019 x86_64 GNU/Linux
root@sos-dom0 ~/video/H264 # acrn-dm -v
DM version is: 0.6-unstable-”2019w07.5.140000p_162” (daily tag:”2019w07.5.140000p”), build by mockbuild@2019-02-15 17:37:59

 

UOS OS Type: Linux

UOS Kernel version: 

root@clr-2664db3a47e34aab850bdea7a528f8a5 ~ # uname -a
Linux clr-2664db3a47e34aab850bdea7a528f8a5 4.19.20-14.iot-lts2018 #1 SMP Fri Feb 15 05:32:27 UTC 2019 x86_64 GNU/Linux

Bug detailed description:
-------------
Video image flicker when play H264 video, other format video can not play

Reproduce steps:
-------------
1. Boot up hypervisor, SOS and UOS
2. execute the following commands in sos or uos
weston --tty=1 -i 0 &
export LIBVA_DRIVER_NAME=iHD 
GST_GL_PLATFORM=egl gst-launch-1.0 filesrc location=ToS_1080p_59.94fps_H264_25000kbps_8bits_noHDR_barcoded-v1511090000.mp4 ! qtdemux ! queue ! h264parse ! msdkh264dec ! waylandsink 
3. observe the sos or uos monitor

Actual result:
-------------
Video image flicker when play H264 video, other format video can not play

Expect result:
-------------

Video image normal when play H264 video, and other format video can  play

 

Rate: 100%

Logs as attached.

mwang106, 2019-02-18:
[External_System_ID] ACRN-2696

Mingyuan18, 2019-03-07:
Can't reproduce this issue in below build:
SOS version: 28100
ACRN TAG: 2019w09.5.140000p

closed 2019-03-07 by Mingyuan18