projectacrn/acrn-hypervisor #2576: [MRB][IO]"ST_PERF_DS3_Resume_to_IVE_Android_OS_UI" test result is not meet PV target(apl_sdc_stable )

lidongwangx, 2019-02-21:
Kernel Version/Android Version | 4.19.19-quilt-2e5dc0ac-00135-gdfb7dfd764a1 (190208T031538Z) / 9
-- | --
AOSP Version | PPR1.181005.003
ABL Version | 1906_GP20
IFWI Version | 3.1.55.2278a
IOC Version | 4.0.14
ACRN-Hypervisor | 2019w05.3.150000p_156
SOS Build Link | 20190211-27700
SOS Version | Installed version: 27700
SOS Kernel Version | 4.19.19-8-.iot-lts2018-sos #1 SMP PREEMPT Mon Feb 11
cAVS FW Version | 9.22.1.3472

Test environment settings
Connect 1 HDMI(on HDMI port2) and 1 eDP panel
Execution steps
Freshly flash the USER image on DUT (Device Under Test) and perform System Setup & Configuration section's steps.
Boot the device to home screen and wait 5 minutes.
Press 'ignition' button let the DUT go to S3 mode and wait about 1 minutes.
open mobile video
Start measurement by pressing the 'ignition' button.
Until the screen lights up and stop video
copy video from mobile to host
execute "python ffmpeg.py -i *.MOV"
calculating time
Expected result
less than 2 seconds
Actual result
more than 2seconds

lidongwangx, 2019-02-21:
[External_System_ID] ACRN-2721

Mingyuan18, 2019-04-17:
Can't reproduce this issue in below build:
ACRN-Hypervisor: 2019w09.4.153000p_162
SOS : 28102
SOS Kernel Version : 4.19.23-24.iot

closed 2019-04-17 by Mingyuan18