projectacrn/acrn-hypervisor #2581: [MRB][IO]There is no Read value output in the log when test KPI "ST_PERF_USB2/3_mass_storage_protocol_Read_Large_File_as_Host"(apl_sdc_stable)

lidongwangx, 2019-02-21:
HW MODEL
BXT-P-GP-MRB-ACRN(D0)
==================================

Kernel Version/Android Version | 4.19.8-quilt-2e5dc0ac-00100-g7b335204cf3e (190109T063124Z) / 9
-- | --
AOSP Version | PPR1.181005.003
ABL Version | 1901_GP20
IFWI Version | 3.1.55.2278a
IOC Version | 4.0.13
ACRN-Hypervisor | 2019w02.2.150000p-141
SOS Build Link | 20190110-27151
SOS Version | Installed version: 27151
SOS Kernel Version | 4.19.8-1901090631.iot-lts2018-sos #1 SMP   PREEMPT Thu Jan 10
cAVS FW   Version | 9.22.1.3472
==================================
SUMMARY
[BXT-P][Gordon_Peak_ACRN]There is no Read value output in the log when test KPI "ST_PERF_USB2/3_mass_storage_protocol_Read_Large_File_as_Host".

TCD LINK:https://docs.google.com/document/d/1GbzQGs49qQS0z_aVFHHnLLQZcwkyNjzU6Rmo6vIb0Io/edit#
==========================================================
EXACT STEPS LEADING TO PROBLEM

Hardware requirement

0 flash disk: SanDisk (CZ60) 64GB
0 flash disk: USCORSAIR GTX USB3.0 128GB
Resource file requirement

Download Iozone
[Precondition]:
1.Disable WLAN/BT/Location.
2.Enable Airplane mode.

[Tools Setup]:

Insert USB2.0 flash disk & USB3.0 flash disk, and erase it from Settings -> Storage -> Portable storage
Push the iozone into DUT /data/ partition.
~$ adb root

~$ adb push iozone /data

~$ adb shell chmod 777 /data/iozone

Get the USB2.0 flash disk & USB3.0 flash disk mount point via command ‘df’, it should be mounted under /mnt/media_rw/ (In my device USB2.0 flash disk mount point is /mnt/media_rw/9A1E-1301, USB3.0 flash disk mount point is /mnt/media_rw/4A04-12E7)
Filesystem               Size     Used     Free   Blksize

/                        3.8G    25.6M     3.8G   4096

……...

/mnt/media_rw/C785-13E2   119.2G    64.0K   119.2G   32768

/mnt/media_rw/9A1E-1301    59.7G    64.0K    59.7G   32768

[Execution Steps]:

Perform System Setup & Configuration section's steps.
Clear the buffers and cache prior to running the test.
root@android:/data# sync && echo 3 > /proc/sys/vm/drop_caches
Create the file that will be used in following tests:
/data/iozone -ec -+n -L64 -S32 -s3G -f /mnt/media_rw/88D0-12E9/iozone.tmp -r512k -i0 -w > /dev/null
Measure Sequential Read (3GB) performances in MB/s.
/data/iozone -ec -+n -L64 -S32 -s3G -f /mnt/media_rw/88D0-12E9/iozone.tmp -w -+N -I -r512k -i1 > /data/seqRead-run.log
Pull out the log files seq-run.log from target.
adb pull /data/seqRead-run.log
Sample Result Format:
KB           reclen  write  rewrite  read    reread  read  write  read  rewrite  read  fwrite frewrite  fread  freread
3145728  512                               39231      0                                                                          
Sequential Read (MB/s) = 39231/1000 = 39.231MB/s
Separate measurement USB2.0 flash disk & USB3.0 flash disk.
Reboot DUT, erase USB2.0 flash disk & USB3.0 flash disk for five measurements.
[Expected Results]: 
There should be a read value output in the seqRead.log.

[Actual Results]: 
There is no read value output in the seqRead.log.

REPRODUCIBILITY
50%
==========================================================
IMPACT ON USER
None
==========================================================
ADDITIONAL INFO
None

lidongwangx, 2019-02-21:
[External_System_ID] ACRN-2710

Mingyuan18, 2019-03-07:
Can't reproduce this issue in below build:
ACRN-Hypervisor : 2019w09.4.153000p_162
SOS: 28102

closed 2019-03-07 by Mingyuan18