projectacrn/acrn-hypervisor #2624: Loading PCI devices with table_count > CONFIG_MAX_MSIX_TABLE_NUM leads to writing outside of struct.

DunderRoffe, 2019-02-26:
When an PCI device is added with an MSI-X table_count larger than CONFIG_MAX_MSIX_TABLE_NUM, the init function in the handler for the MIS-X capability (vmsix_init in msix.c) will write outside of the tables array in the pci_msix struct when initializing said array.

The reason for this is that the size of the array is set to CONFIG_MAX_MSIX_TABLE_NUM but there are no runtime checks that make sure that this is not violated. So when a PCI device is detected during boot with a table_count (which is read from the PCI device itself) larger than CONFIG_MAX_MSIX_TABLE_NUM the initialization will write outside of the array.

From vmsix_init in msix.c
```c
for (i = 0U; i < msix->table_count; i++) {
			msix->tables[i].vector_control = PCIM_MSIX_VCTRL_MASK;
			msix->tables[i].addr = 0U;
			msix->tables[i].data = 0U;
}
```

The definition of the pci_msix struct from vpci.h:
```c
/* MSI-X capability structure */
struct pci_msix {
	struct msix_table_entry tables[CONFIG_MAX_MSIX_TABLE_NUM];
	uint64_t  mmio_gpa;
	uint64_t  mmio_hva;
	uint64_t  mmio_size;
	uint64_t  intercepted_gpa;
	uint64_t  intercepted_size;
	uint32_t  capoff;
	uint32_t  caplen;
	uint32_t  table_bar;
	uint32_t  table_offset;
	uint32_t  table_count;
};
```

Currently there is no indication when this happens, which in my case meant 1 week of debugging the hypervisor to understand what has happened. In my perticular case the vpci device which got initialized after the pci device with too high table_count all of a sudden had the value 0 for all of the fields in its struct. This included the vpci which caused page fault when starting the SOS VM.

Mingyuan18, 2019-02-27:
[External_System_ID] ACRN-2826

DunderRoffe, 2019-03-11:
This should not be a problem anymore since #2616 was merged

closed 2019-03-11 by DunderRoffe