projectacrn/acrn-hypervisor #2659: Use command 'launch_uos.sh -V 2 -d' failed

Ragnarliu, 2019-03-01:
Use command 'launch_uos.sh -V 2 -d' failed.
sh launch_uos.sh -V 2 -d 
10:49:33.755 INFO launch_uos.sh: option requires an argument – d
Use luanch_uos.sh like that ./launch_uos.sh -V <#>
The option -V means the UOSs group to be launched by vsbl as below
-V 1 means just launching 1 clearlinux UOS
-V 2 means just launching 1 android UOS
-V 3 means launching 1 clearlinux UOS + 1 android UOS
-V 4 means launching 2 clearlinux UOSs
-V 5 means just launching 1 alios UOS
-V 6 means auto check android/linux/alios UOS; if exist, launch it
-C means launching acrn-dm in runC container for QoS
[38;5;39mroot[0m@[38;5;208msos-dom0 [38;5;39m~ [38;5;39m# [0;0m

Mingyuan18, 2019-03-01:
[External_System_ID] ACRN-2859

Mingyuan18, 2019-06-05:
Modify the launch script and fix the issue.

closed 2019-06-05 by Mingyuan18