projectacrn/acrn-hypervisor #2661: The "Using SBL on UP2 Board" tutorial is broken with the latest ACRN code

gvancuts, 2019-03-01:
Disclaimer: I have not tested this but I nevertheless prefer to capture what I believe is broken today in that tutorial.

The "Using SBL on UP2 Board" tutorial uses a [script](https://raw.githubusercontent.com/projectacrn/acrn-hypervisor/master/doc/tutorials/create-up2-images.sh) from the repo to build the SOS and UOS images. A recent commit (see https://github.com/projectacrn/acrn-hypervisor/commit/f9b5e21bdc325a57a253001685d26174984d212c) changed the name of the UP2 board from `up2` to `apl-up2`. That script needs to be updated to use the new name.

gvancuts, 2019-03-01:
@ttzeng could you take a look at this?

ttzeng, 2019-03-02:
@gvancuts you are right, errors are thrown while building the SOS image with the ``create-up2-images.sh`` script if ``--acrn-code-path`` argument is given.

dbkinder, 2019-03-22:
Resolved.

closed 2019-03-22 by dbkinder