projectacrn/acrn-hypervisor #2683: Calling `reboot` from the ACRN console triggers a crash

gvancuts, 2019-03-05:
Environment:
* UP2 board, [UEFI bios version 4.0](https://downloads.up-community.org/download/up-squared-uefi-bios-v4-0/)
* ACRN: HV version 0.7-unstable-2019-03-01 (from Clear Linux 28100)
* Using `uart=bdf@0:18.0 vuart=ttyS0@irq4`

Calling `reboot` from the ACRN console triggers a crash, see error message below:
```
ACRN:\>reboot
trigger crash, divide by 0 ...

================================================================================
=
= Unhandled exception: 0 (Divide Error)

Host Registers:
=  Vector=0x0000000000000000  RIP=0x000000005B229957
=     RAX=0x0000000000000001  RBX=0x0000000000000028  RCX=0x0000000000000000
=     RDX=0x0000000091526005  RDI=0x000000000000000D  RSI=0x0000000091526000
=     RSP=0x0000000065082650  RBP=0x0000000065082808  RBX=0x0000000000000028
=      R8=0x0000000091526000   R9=0x0000000091526000  R10=0x0000000000000000
=     R11=0x0000000000000000  R12=0x00000000650826F0  R13=0x000000000000000C
=  RFLAGS=0x0000000000010212  R14=0x0000000065082E5B  R15=0x00000018D97EF8AC
= ERRCODE=0x0000000000000000   CS=0x0000000000000008   SS=0x00000000650826E8

================================================================================

Host Stack: CPU_ID = 0
addr(0x65082650)        0x0000000065082808  0x0000000091526000  0x000000000000000d  0x0000000091526000
addr(0x65082670)        0x0000000091526000  0x0000000000000000  0x0000000000000000  0x00000000650826f0
addr(0x65082690)        0x000000000000000c  0x0000000065082e5b  0x00000018d97ef8ac  0x0000000000000000
addr(0x650826b0)        0x0000000000000000  0x000000005b229957  0x0000000000000008  0x0000000000010212
addr(0x650826d0)        0x00000000650826e8  0x0000000000000010  0x000000005b22994d  0x0000000000000000
addr(0x650826f0)        0x2072656767697274  0x64202c6873617263  0x7962206564697669  0x0a0d2e2e2e203020
addr(0x65082710)        0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000
addr(0x65082730)        0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000
addr(0x65082750)        0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000
addr(0x65082770)        0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000
addr(0x65082790)        0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000
addr(0x650827b0)        0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000
addr(0x650827d0)        0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000
addr(0x650827f0)        0xeda79ac889b30217  0x000000005b2464c0  0x0000000000000001  0x0000000065082ee8
addr(0x65082810)        0x000000005b229757  0x0000000065082e5b  0x0000000065082e68  0x0000000065082e6a
addr(0x65082830)        0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000

Host Call Trace:
----> 0x000000005b229757
----> 0x000000005b22b1c1
----> 0x000000005b22b2af
----> 0x000000005b207c6a
----> 0x000000005b21a344
----> 0x000000005b21a39c
----> 0x000000005b21c484

================================================================================

Guest Registers:
=       VM ID 0 ==== vCPU ID 0 ===  pCPU ID 0 ====world 0 =============
=       RIP=0xffffffffa2bf2259  RSP=0xffffffffa3603d68 RFLAGS=0x0000000000000093
=       CR0=0x0000000080050033  CR2=0x00005621df88e0a8  CR3=0x000000006a613000
=       RAX=0x0000000000000000  RBX=0xffff8ac77a8958cc  RCX=0x000000000000009c
=       RDX=0x0000000000000419  RDI=0xffff8ac77a8958cc  RSI=0xffffffffa344af13
=       RBP=0xffffffffa3603d68  R8=0x0000000000000000  R9=0x0000000000000002
=       R10=0x00000018d9765b9c  R11=0xffff8ac77ba206a8  R12=0xffff8ac77a895800
=       R13=0x0000000000000001  R14=0xffff8ac77aa4e000  R15=0x0000000000000003

Guest Stack:
Dump stack for vcpu 0, from gva 0xffffffffa3603d68
guest_rsp(0xffffffffa3603d68):  0xffffffffa3603d90  0xffffffffa251c66a  0xffff8ac77a895800  0xffff8ac77a8958cc
guest_rsp(0xffffffffa3603d88):  0x00000000ffffffea  0xffffffffa3603de0  0xffffffffa251ca51  0xffffffffa36c7ac0
guest_rsp(0xffffffffa3603da8):  0x00000000a3603db8  0xf310644013199200  0x0000000000000003  0xffff8ac77aa4e000
guest_rsp(0xffffffffa3603dc8):  0xffffffffa36c7ac0  0xffffffffa36c7bf8  0xffffffffa36c7be0  0xffffffffa3603e30
guest_rsp(0xffffffffa3603de8):  0xffffffffa295b038  0xffffffffa3650fe8  0xffff8ac700a4e000  0x000000122abd4f3b
guest_rsp(0xffffffffa3603e08):  0x0000000000000000  0xffffffffa37c30b0  0xffff8ac77aa4e000  0xffffffffa36c7ac0
guest_rsp(0xffffffffa3603e28):  0x0000000000000003  0xffffffffa3603e40  0xffffffffa295b357  0xffffffffa3603e50
guest_rsp(0xffffffffa3603e48):  0xffffffffa20d6713  0xffffffffa3603e98  0xffffffffa20d6b4a  0x00ffffffa20ccf8f
guest_rsp(0xffffffffa3603e68):  0xf310644013199200  0x00000000000000c5  0xffff8ac77fff80c0  0xffffffffa38c0920
guest_rsp(0xffffffffa3603e88):  0xffffffffa38c72e0  0x0000000072c0b878  0xffffffffa3603ec0  0xffffffffa20d6d93
guest_rsp(0xffffffffa3603ea8):  0xf310644013199200  0x1deab8dd7f2f3200  0x0000000000000002  0xffffffffa3603ed8
guest_rsp(0xffffffffa3603ec8):  0xffffffffa2be85ac  0xffffffffffffffff  0xffffffffa3603f20  0xffffffffa38120ab
guest_rsp(0xffffffffa3603ee8):  0xffffffffa38c72e0  0x1deab8dd7f2f3200  0x0000000000000000  0x0000000000001000
guest_rsp(0xffffffffa3603f08):  0x000000007474e218  0x0000000000000000  0x0000000076bae0d8  0xffffffffa3603f30
guest_rsp(0xffffffffa3603f28):  0xffffffffa3811425  0xffffffffa3603f48  0xffffffffa3811496  0x0000000000000000
guest_rsp(0xffffffffa3603f48):  0x0000000000000000  0xffffffffa20000d4  0x0000000000000000  0x0000000000000000

Guest Call Trace: **************************************
Maybe the call trace is not accurate, pls check stack!!
BP_GVA(0xffffffffa3603d68) RIP=0xffffffffa3603d90
BP_GVA(0xffffffffa3603d90) RIP=0xffffffffa3603de0
BP_GVA(0xffffffffa3603de0) RIP=0xffffffffa3603e30
BP_GVA(0xffffffffa3603e30) RIP=0xffffffffa3603e40
BP_GVA(0xffffffffa3603e40) RIP=0xffffffffa3603e50
BP_GVA(0xffffffffa3603e50) RIP=0xffffffffa3603e98
BP_GVA(0xffffffffa3603e98) RIP=0xffffffffa3603ec0
BP_GVA(0xffffffffa3603ec0) RIP=0xffffffffa3603ed8
BP_GVA(0xffffffffa3603ed8) RIP=0xffffffffa3603f20
BP_GVA(0xffffffffa3603f20) RIP=0xffffffffa3603f30
BP_GVA(0xffffffffa3603f30) RIP=0xffffffffa3603f48
BP_GVA(0xffffffffa3603f48) RIP=0x0000000000000000
```

gvancuts, 2019-03-05:
OK, I see now that this is somehow the intended result... this is what the `reboot` command does: https://github.com/projectacrn/acrn-hypervisor/blob/master/hypervisor/debug/shell.c#L1247-L1261

This is rather confusing, could we rename this command to something more meaningful?

gvancuts, 2019-03-19:
@dongyaozu , what is your recommendation? Should we rename the `reboot` command into something more meaningful, e.g. `trigger_crash`?

yifanddd, 2019-04-12:
[External_System_ID] ACRN-3375

fuzhongl, 2019-05-28:
@gvancuts As it works as design, Can we close this ticket?
Thanks!

gvancuts, 2019-05-28:
We should do at least one of two things before closing this:
1. Rename the hypervisor shell function to `trigger_crash` (or something similar)
2. Fix the shell command so that it *really* reboots the platform (as the name implies)

We could even do both really if we feel that having both commands (`reboot` and `trigger_crash` is useful)

gvancuts, 2019-06-04:
The `reboot` shell command will be removed, tracking this in #3210 and hence closing this issue.

closed 2019-06-04 by gvancuts