projectacrn/acrn-hypervisor #2704: Possible memory leak issues

Mingyuan18, 2019-03-08:
Possible memory leak issues

Mingyuan18, 2019-03-08:
[External_System_ID] ACRN-2935

yonghuah, 2019-03-08:
/acrn-hypervisor/devicemodel/hw/platform*/usb_pmapper.c*
Possible memory leak. Dynamic memory stored in 'req' allocated through function 'usb_dev_alloc_req' 

/acrn-hypervisor/devicemodel/hw/pci/passthrough.c
Possible memory leak. Dynamic memory stored in 'ptdev' allocated through function 'calloc' 
 
/acrn-hypervisor/devicemodel/hw*/pci/xhci.c*
Possible memory leak. Dynamic memory stored in 'xdev->devices' allocated through function 'pci_xhci_parse_opts' 

acrn-hypervisor/devicemodel/hw/pci/xhci.c
Possible memory leak. Dynamic memory stored in 'xdev->devices' allocated through function 'calloc'
 
acrn-hypervisor/devicemodel/hw*/pci/xhci.c
Possible memory leak. Dynamic memory stored in 'xdev->slots' allocated through function 'pci_xhci_parse_opts' 
 
acrn-hypervisor/devicemodel/hw/platform/usb_pmapper.c
Possible memory leak. Dynamic memory stored in 'req->buffer' allocated through function 'usb_dev_alloc_req' 
 
acrn-hypervisor/devicemodel/core/sw_load_elf.c
Possible memory leak. Dynamic memory stored in 'elf32_phdr' allocated through function 'calloc' 
 
/acrn-hypervisor/devicemodel/hw/pci/virtio*/virtio_net.c*
Possible memory leak. Dynamic memory stored in 'net' allocated through function 'calloc'

Mingyuan18, 2019-03-21:
This issue was fixed and no regression issue found in test.

closed 2019-03-21 by Mingyuan18