projectacrn/acrn-hypervisor #2734: Buffer Overflow Vulnerability in 'acrnctl'

taoyuhong, 2019-03-12:
While performing introspected fuzzing on the acrnctl compiled binaries, multiple crash conditions were encountered that resulted in Segmentation Fault conditions that could be exploited by an attacker. Buffer overflows can often be triggered by malformed inputs If this overwrites adjacent data or executable code, this may result in erratic program behavior, including memory access errors, incorrect results, and crashes.

Steps to Reproduce:
 1. sudo afl-fuzz -i in/ -o out/ – ./acrnctl add @@
 2. gdb ./acrnctl
 3. set break point
 4. r add crashfile (located in crashes folder)
 5. notice that application crashes with segmentation fault.

Business Risk
 When this condition is invoked, an attacker may be able to execute arbitrary code taking control of the endpoint system,altering the intended control flow, or reading potentially sensitive information. Failed attempts will usually cause the system or program to crash.

Recommendation
 Verify that any buffer allocations are as large as you specify before performing operations. Verify buffer boundaries if accessing the buffer in a loop, and make sure the code logic is not in danger of writing past the allocated space. If necessary, truncate all input strings to a reasonable length prior to passing them to the copy and concatenation functions

[External_System_ID] ACRN-2389

wenlingz, 2019-03-13:
[External_System_ID] ACRN-2389

Mingyuan18, 2019-03-18:
Can't reproduce this issue in below build, So close it.
ACRN tag: 2019w11.3.153000p
SOS:28100

closed 2019-03-18 by Mingyuan18