projectacrn/acrn-hypervisor #2737: Doc: GSG need update for Build ACRN from Source section

Mingyuan18, 2019-03-12:
GSG for Build hypervisor section is out of date, need update.

Mingyuan18, 2019-03-12:
[External_System_ID] ACRN-2969

gvancuts, 2019-03-29:
@Mingyuan18 , can you be more specific as to what is out of date? (I must confess I haven't tried with Ubuntu 16.04 for a while so if you can give me some pointers as to the problems you encountered it would be helpful)

wenlingz, 2019-04-24:
Please refer the fixed method:

run these command:

apt-get install openwince-include 

cp /usr/include/openwince/arm/sa11x0/gpio.h  /usr/include/linux/

fixed.

gvancuts, 2019-04-24:
There are a number of issues with the current set of instructions for Ubuntu 16.04.
* Missing instructions on how to upgrade `binutils` to >=2.26
* Missing the `pkg-config` package
* Missing header file (for GPIO)

@wenlingz , I followed your instructions above to add the missing `gpio.h` header file but I'm now hitting the following error message:
```
In file included from hw/pci/virtio/virtio_gpio.c:20:0:
/usr/include/linux/gpio.h:42:10: fatal error: common.h: No such file or directory
 #include <common.h>
          ^~~~~~~~~~
compilation terminated.
```

gvancuts, 2019-04-24:
I did a `cp /usr/include/openwince/common.h /usr/include/` to get around this new error and I am now hitting the following error:
```
cc -g -O0 -std=gnu11 -D_GNU_SOURCE -DNO_OPENSSL -m64 -Wall -ffunction-sections -Werror -O2 -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -fno-strict-aliasing -fpie -I/root/acrn/devicemodel/include -I/root/acrn/devicemodel/include/public -I/root/acrn/build/devicemodel/include -I/root/acrn/build/tools -fstack-protector-strong -DDM_DEBUG -c hw/pci/virtio/virtio_gpio.c -o /root/acrn/build/devicemodel/hw/pci/virtio/virtio_gpio.o
hw/pci/virtio/virtio_gpio.c: In function 'native_gpio_update_line_info':
hw/pci/virtio/virtio_gpio.c:298:23: error: storage size of 'info' isn't known
  struct gpioline_info info;
                       ^~~~
hw/pci/virtio/virtio_gpio.c:303:29: error: 'GPIO_GET_LINEINFO_IOCTL' undeclared (first use in this function)
  rc = ioctl(line->chip->fd, GPIO_GET_LINEINFO_IOCTL, &info);
                             ^~~~~~~~~~~~~~~~~~~~~~~
hw/pci/virtio/virtio_gpio.c:303:29: note: each undeclared identifier is reported only once for each function it appears in
hw/pci/virtio/virtio_gpio.c:310:28: error: 'GPIOLINE_FLAG_KERNEL' undeclared (first use in this function)
  line->busy = info.flags & GPIOLINE_FLAG_KERNEL;
                            ^~~~~~~~~~~~~~~~~~~~
hw/pci/virtio/virtio_gpio.c:320:27: error: 'GPIOLINE_FLAG_IS_OUT' undeclared (first use in this function); did you mean 'GPIOLINE_FLAG_KERNEL'?
  line->dir = info.flags & GPIOLINE_FLAG_IS_OUT ? 0 : 1;
                           ^~~~~~~~~~~~~~~~~~~~
                           GPIOLINE_FLAG_KERNEL
hw/pci/virtio/virtio_gpio.c:298:23: error: unused variable 'info' [-Werror=unused-variable]
  struct gpioline_info info;
                       ^~~~
hw/pci/virtio/virtio_gpio.c: In function 'native_gpio_open_line':
hw/pci/virtio/virtio_gpio.c:356:28: error: storage size of 'req' isn't known
  struct gpiohandle_request req;
                            ^~~
hw/pci/virtio/virtio_gpio.c:373:15: error: 'GPIOHANDLE_REQUEST_OUTPUT' undeclared (first use in this function)
   if (flags & GPIOHANDLE_REQUEST_OUTPUT)
               ^~~~~~~~~~~~~~~~~~~~~~~~~
hw/pci/virtio/virtio_gpio.c:376:29: error: 'GPIO_GET_LINEHANDLE_IOCTL' undeclared (first use in this function)
  rc = ioctl(line->chip->fd, GPIO_GET_LINEHANDLE_IOCTL, &req);
                             ^~~~~~~~~~~~~~~~~~~~~~~~~
hw/pci/virtio/virtio_gpio.c:388:22: error: 'GPIOHANDLE_REQUEST_INPUT' undeclared (first use in this function); did you mean 'GPIOHANDLE_REQUEST_OUTPUT'?
   } else if (flags & GPIOHANDLE_REQUEST_INPUT)
                      ^~~~~~~~~~~~~~~~~~~~~~~~
                      GPIOHANDLE_REQUEST_OUTPUT
hw/pci/virtio/virtio_gpio.c:356:28: error: unused variable 'req' [-Werror=unused-variable]
  struct gpiohandle_request req;
                            ^~~
hw/pci/virtio/virtio_gpio.c: In function 'gpio_set_value':
hw/pci/virtio/virtio_gpio.c:401:25: error: storage size of 'data' isn't known
  struct gpiohandle_data data;
                         ^~~~
hw/pci/virtio/virtio_gpio.c:413:23: error: 'GPIOHANDLE_SET_LINE_VALUES_IOCTL' undeclared (first use in this function)
  rc = ioctl(line->fd, GPIOHANDLE_SET_LINE_VALUES_IOCTL, &data);
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
hw/pci/virtio/virtio_gpio.c:401:25: error: unused variable 'data' [-Werror=unused-variable]
  struct gpiohandle_data data;
                         ^~~~
hw/pci/virtio/virtio_gpio.c: In function 'gpio_get_value':
hw/pci/virtio/virtio_gpio.c:427:25: error: storage size of 'data' isn't known
  struct gpiohandle_data data;
                         ^~~~
hw/pci/virtio/virtio_gpio.c:453:17: error: 'GPIOHANDLE_GET_LINE_VALUES_IOCTL' undeclared (first use in this function)
  rc = ioctl(fd, GPIOHANDLE_GET_LINE_VALUES_IOCTL, &data);
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
hw/pci/virtio/virtio_gpio.c:427:25: error: unused variable 'data' [-Werror=unused-variable]
  struct gpiohandle_data data;
                         ^~~~
hw/pci/virtio/virtio_gpio.c: In function 'gpio_set_direction_input':
hw/pci/virtio/virtio_gpio.c:468:37: error: 'GPIOHANDLE_REQUEST_INPUT' undeclared (first use in this function)
  return native_gpio_open_line(line, GPIOHANDLE_REQUEST_INPUT, 0);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~
hw/pci/virtio/virtio_gpio.c: In function 'gpio_set_direction_output':
hw/pci/virtio/virtio_gpio.c:478:37: error: 'GPIOHANDLE_REQUEST_OUTPUT' undeclared (first use in this function)
  return native_gpio_open_line(line, GPIOHANDLE_REQUEST_OUTPUT, value);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
hw/pci/virtio/virtio_gpio.c: In function 'gpio_set_config':
hw/pci/virtio/virtio_gpio.c:507:12: error: 'GPIOLINE_FLAG_OPEN_DRAIN' undeclared (first use in this function); did you mean 'PIN_CONFIG_DRIVE_OPEN_DRAIN'?
   config = GPIOLINE_FLAG_OPEN_DRAIN;
            ^~~~~~~~~~~~~~~~~~~~~~~~
[... 44 lines cut ...]
hw/pci/virtio/virtio_gpio.c:1098:30: error: 'GPIO_GET_LINEEVENT_IOCTL' undeclared (first use in this function)
  err = ioctl(line->chip->fd, GPIO_GET_LINEEVENT_IOCTL, &req);
                              ^~~~~~~~~~~~~~~~~~~~~~~~
hw/pci/virtio/virtio_gpio.c:1053:27: error: unused variable 'req' [-Werror=unused-variable]
  struct gpioevent_request req;
                           ^~~
hw/pci/virtio/virtio_gpio.c: In function 'gpio_get_value':
[... cut ...]
hw/pci/virtio/virtio_gpio.c: In function 'gpio_set_direction_input':
hw/pci/virtio/virtio_gpio.c:469:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: all warnings being treated as errors
Makefile:219: recipe for target '/root/acrn/build/devicemodel/hw/pci/virtio/virtio_gpio.o' failed
make[1]: *** [/root/acrn/build/devicemodel/hw/pci/virtio/virtio_gpio.o] Error 1
make[1]: Leaving directory '/root/acrn/devicemodel'
Makefile:46: recipe for target 'devicemodel' failed
make: *** [devicemodel] Error 2
```

gvancuts, 2019-04-24:
The installation of the `openwince-include` and copying of header files around feels hacky to me. Is it guaranteed that the header files are of the right version? If not, wouldn't it be better to actually ship the correct version of the header file with our source code?

Mingyuan18, 2019-05-08:
Fixed now.

closed 2019-05-08 by Mingyuan18