projectacrn/acrn-hypervisor #2782: Wifi Firmware initialization failed happened on Reboot

kiendinh, 2019-03-15:
We encountered the same issue #2328 on NUC7i7DNHE (KBL - ACRN v0.4). Is the patch also applied for platforms other than APL?

binbinwu1, 2019-03-18:
What's the WIFI chip used on NUC7i7DNHE?

kiendinh, 2019-03-18:
It’s the dual band wireless 8265.

binbinwu1, 2019-03-21:
@kiendinh As discussed, please unload the wifi driver in UOS first before UOS reboot.

Mingyuan18, 2019-03-21:
@kiendinh : If "unload the wifi driver in UOS first before UOS reboot" can solve your problem, please close this issue, thank you！

gvancuts, 2019-03-21:
> @kiendinh : If "unload the wifi driver in UOS first before UOS reboot" can solve your problem, please close this issue, thank you！

While unloading the driver gives us a good workaround, it feels to me that there is something still in the stack that ought to be fixed. Or am I wrong?

kiendinh, 2019-03-22:
Agreed with @gvancuts. In fact there are many situations that the VM gets reboot without unbinding the drivers. It could be ideal if ACRN detects the status of the VM and returns the resource on its exit.

binbinwu1, 2019-03-22:
@gvancuts @kiendinh, basically, when a device is passed-though to UOS, the software stack of UOS need to reset the status of the device when device. The thing is not easily done in SOS.
1. SOS may not have the driver of the device.
2. Even if SOS has the driver of the device, if the device is not de-initialzed by UOS, it is probably fail to reset the device in SOS.
The best way to resolve this kind of issue is that the device driver should have a way to reset/re-initialize the device during reboot. For example, reset the device status before init the device. 
But this kind of change is out of scope that ACRN can do.

gvancuts, 2019-03-22:
>    The best way to resolve this kind of issue is that the device driver should have a way to reset/re-initialize the device during reboot. For example, reset the device status before init the device.

Thanks for these additional details @binbinwu1 , does that mean that the UOS does not do a driver reset when rebooting? Is this expected? (sorry if this is a silly question, I do not know enough about these low-level details)

binbinwu1, 2019-03-22:
@gvancuts linux kernel provides the interface, which has a shutdown() callback function for the driver to implement. And the callback function will be called during shutdown or reboot if implemented. However, it is up to a driver to implement shutdown() callback, and not all drivers implement that. 
Unfortunately, I checked that Intel wireless driver doesn't implement the  shutdown() callback.
The workaround works because when you unload the driver in UOS, another callback function remove(), which is implemented, will be called to do the thing of cleanup.

binbinwu1, 2019-03-22:
@gvancuts BTW, there is no such issue on native linux/android environment is because during reboot, there will be power cycling. However in virtualization platform, there is no such power cycling when you just reboot UOS.

gvancuts, 2019-03-22:
> @gvancuts BTW, there is no such issue on native linux/android environment is because during reboot, there will be power cycling. However in virtualization platform, there is no such power cycling when you just reboot UOS.

Does that mean you'd expect to see the same problem if we were to run the same UOS on Xen or KVM and reboot it?

gvancuts, 2019-03-22:
> @gvancuts linux kernel provides the interface, which has a shutdown() callback function for the driver to implement. And the callback function will be called during shutdown or reboot if implemented. However, it is up to a driver to implement shutdown() callback, and not all drivers implement that.
> Unfortunately, I checked that Intel wireless driver doesn't implement the shutdown() callback.
> The workaround works because when you unload the driver in UOS, another callback function remove(), which is implemented, will be called to do the things of cleanup.

This is really useful background info, thanks!!

binbinwu1, 2019-03-25:
> Does that mean you'd expect to see the same problem if we were to run the same UOS on Xen or KVM and reboot it?

Didn't try on KVM or Xen.

kiendinh, 2019-03-25:
I'm a little confused about the work around. Since acrn-dm is the one who passthroughs the device to a UOS, in general it should be the one who is able to "reset" the device. My counterparts have reported that they could reset by adding reset to the passthru option like:

-s 9,passthru,1/00/0,reset \

where 1:00.0 is the wifi device in the NUC7i7DNHE. Since then the reboot worked fine.

So maybe it is already supported by acrn. We just need to add the info to the document so that other users can be aware of.

binbinwu1, 2019-03-25:
> I'm a little confused about the work around. Since acrn-dm is the one who passthroughs the device to a UOS, in general it should be the one who is able to "reset" the device. My counterparts have [...]

Forgot this option...
Some devices allow an individual function to be reset without affecting other functions in the same device. For devices that have this support, a file named reset will be present in sysfs.  Writing 1 to this file will perform reset.
But I thought this will need the driver for the device loaded. Did you mean it even works when the device is binding to driver pci-stub?

kiendinh, 2019-03-25:
Yes, so in the launch script, it does:

...
echo "8086 24fd" > /sys/bus/pci/drivers/pci-stub/new_id
echo "0000 01:00.0" > /sys/bus/pci/devices/0000:01:00.0/driver/unbind
echo "0000:01:00.0" > /sys/bus/pci/drivers/pci-stub/bind
...

before passing thru the device with acrn-dm
...
-s 9,passthru,1/00/0,reset \
...

binbinwu1, 2019-03-25:
I checked the kernel code when interact with "/sys/bus/pci/devices/.../reset", it will try several way to reset the device, and some of them doesn't need the support from device driver. I think the wifi device should be able to support FLR (function level reset), which is one of the way the PCI subsystem tries to reset the device via device config space.

kiendinh, 2019-03-26:
@binbinwu1 Thank you for the confirmation. So how does it work? acrn-dm monitors the VM, and on vm-exit it will trigger the FLR if the "reset" option is on? In the future, when ACRN supports container-type VMs (ie Kata), there maybe more and more passthru requirements so that a VM can have a better performance with direct HW access. The device needs to be release for other VMs to use when the current VM is done or closed - and I think that would be better for the acrn-dm to support the feature. It would be difficult for a VM - container - to control a specific device it doesn't know until runtime.

yifanddd, 2019-04-12:
[External_System_ID] ACRN-3374

fuzhongl, 2019-05-14:
@kiendinh Can this issue be closed?
Thanks!

fuzhongl, 2019-05-28:
@kiendinh 
Any more concern, please update the comments. If no more feedback within two weeks, this ticket will be closed by default. 
Appreciate for your understanding.

Mingyuan18, 2019-06-11:
Close this issue due to no any feedback.

closed 2019-06-11 by Mingyuan18