projectacrn/acrn-hypervisor #2793: Hard to override EFI loader

rossburton, 2019-03-15:
In acrn-hypervisor up to 0.6 it was possible to override the EFI loader path in the `.config`, made easy for distro packages by writing a `.config` with just that value and then running `make oldconfig`.

0.7 removed that and now the loader path is a `#define` that needs to be patched.  Please consider either reintroducing the Kconfig, or if Kconfig is being phased out then let me pass `-DDEFAULT_UEFI_OS_LOADER_NAME=/some/path` in CPPFLAGS by guarding the `#define` with `#ifndef`.

Mingyuan18, 2019-03-18:
@rossburton  , We will solve this issue with #2745, and please give suggestions about configurations below #2745, thank you!

gvancuts, 2019-03-18:
hi @Mingyuan18 , I fail to see how #2745 is relevant here, can you clarify?

Mingyuan18, 2019-03-18:
@gvancuts Sorry, It should be #2794.

rossburton, 2019-03-18:
The EFI loader configuration *was in the kconfig* in 0.7 but has been removed since (commit 334382f9ef1f0564598cd2f96ad22d1a7f8b4c1c).  If you're planning on adding it back then fine, #2794 needs to be fixed, but with current master it isn't relevant.

I'm currently using this patch so I can at least pass the value in `$CPPFLAGS`:

```
--- a/efi-stub/boot.c
+++ b/efi-stub/boot.c
@@ -336,7 +336,10 @@ static inline EFI_STATUS isspace(CHAR8 ch)
     return ((uint8_t)ch <= ' ');
 }

+#ifndef DEFAULT_UEFI_OS_LOADER_NAME
 #define DEFAULT_UEFI_OS_LOADER_NAME "\\EFI\\org.clearlinux\\bootloaderx64.efi"
+#endif
```

Mingyuan18, 2019-03-28:
@rossburton , Can you close this issue? Because issue  #2794 has already merged the patch

yifanddd, 2019-04-12:
[External_System_ID] ACRN-3372

yifanddd, 2019-04-12:
[External_System_ID] ACRN-3373

fuzhongl, 2019-05-28:
@rossburton 
Any more concern, please update the comments. If no more feedback within two weeks, this ticket will be closed by default. 
Appreciate for your understanding.

rossburton, 2019-06-02:
Still difficult.  See https://github.com/projectacrn/acrn-hypervisor/blob/master/efi-stub/boot.c#L350

There's no easy way to override that without a patch.

fuzhongl, 2019-06-12:
> Still difficult. See https://github.com/projectacrn/acrn-hypervisor/blob/master/efi-stub/boot.c#L350
> 
> There's no easy way to override that without a patch.

@rossburton  Could you please raise a patch for this issue?
Thanks!

rossburton, 2019-06-17:
The patch we've got just goes in and changes the value.  There should be a way to set it from an environment variable of some sort.

fuzhongl, 2019-10-29:
@rossburton Sorry for update so later.
Could you please use   bootloader option of efibootmgr to override EFI loader? Thanks!
$ sudo efibootmgr -c -l "\EFI\acrn\acrn.efi" -d /dev/sda -p 1 -L "ACRN NUC Hypervisor" \
      -u "bootloader=\EFI\your_path\bootloaderx64.efi uart=port@0x3f8"

gvancuts, 2019-10-31:
We know this can be done during the installation phase, but we (also) need a way to configure this at build time. The reason why we removed this option completely escapes me, just as I really don't understand why there seems to be a strong reluctance to bring this back. Can you enlighten me on this?

Let me ask differently perhaps, if I prepare a PR to bring this back, are you open to taking it into our codebase?

fuzhongl, 2019-10-31:
@gvancuts I will ask our arch to answer this question since he has some concern about it.
Thanks!

fuzhongl, 2019-11-04:
@gvancuts  Double checked with our ARCH; It is fine to modify the patch of bootloader; please help to raise PR.
Thanks!

gvancuts, 2019-11-07:
> @gvancuts Double checked with our ARCH; It is fine to modify the patch of bootloader; please help to raise PR.
> Thanks!

Thanks @fuzhongl , I have created PR #4019 to try address this. Can you help review it?

closed 2020-05-29 by fuzhongl