projectacrn/acrn-hypervisor #2794: Difficult to use make oldconfig

rossburton, 2019-03-15:
With 0.7 its harder than before to actually manipulate the kconfig without using menuconfig, which isn't an option for distribution-built packages that need to be customised.

Our EFI boot loader isn't at `\\EFI\\org.clearlinux\\bootloaderx64.efi`, which is the default.  Thanks to some previous cleanups (see #2371) we can now use `make oldconfig` as is typical:

```
$ cat <<EOF >hypervisor/build/.config
CONFIG_BOARD="NUC7i7BNH"
CONFIG_UEFI_OS_LOADER_NAME="foobar"
EOF
$ make -C hypervisor oldconfig
```

As expected this populates the rest of the `.config` with all the missing values, preserving the values that I'd already set.  However, this isn't then able to compile:

```
$ make -C hypervisor
...
cc  -Iinclude  -Iinclude/lib  -Iinclude/lib/crypto  -Iinclude/common  -Iinclude/arch/x86  -Iinclude/arch/x86/boot  -Iinclude/arch/x86/guest  -Iinclude/debug  -Iinclude/public  -Iinclude/dm  -Ibsp/include  -Iboot/include  -I/home/ross/Programming/acrn-hypervisor/hypervisor/build/include  -Iarch/x86/configs/NUC7i7BNH  -Ibsp/include/uefi -I. -c  -include /home/ross/Programming/acrn-hypervisor/hypervisor/build/include/config.h -Wall -W -ffunction-sections -fdata-sections -fshort-wchar -ffreestanding -fsigned-char -m64 -mno-mmx -mno-sse -mno-sse2 -mno-80387 -mno-fp-ret-in-387 -mno-red-zone -nostdinc -nostdlib -fno-common -Werror -fpie -fstack-protector-strong -DSTACK_PROTECTOR -DHV_DEBUG -DPROFILING_ON -fno-omit-frame-pointer  -gdwarf-2 arch/x86/configs/vm_config.c -o /home/ross/Programming/acrn-hypervisor/hypervisor/build/arch/x86/configs/vm_config.o -MMD -MT /home/ross/Programming/acrn-hypervisor/hypervisor/build/arch/x86/configs/vm_config.o
arch/x86/configs/vm_config.c:15:20: fatal error: sos_vm.h: No such file or directory
 #include <sos_vm.h>
                    ^
```

The important part is  `-Iarch/x86/configs/NUC7i7BNH`, because the include directory is actually called `arch/x86/configs/nuc7i7bnh`.  I can't use the lowercase form when running `make oldconfig` because then it doesn't match the `CONFIG_BOARD` in the Kconfig files.

The only way to build the hypervisor using a custom `.config` is to seed the file with `CONFIG_BOARD="NUC7i7BNH"` but also redundantly pass `BOARD= nuc7i7bnh` to `make`.

I believe that `BOARD` should only be needed when running `defconfig` or `oldconfig`, and that the `CONFIG_BOARD` name in the Kconfig should be canonical: i.e. the include directories renamed.

Mingyuan18, 2019-03-18:
[External_System_ID] ACRN-3044

NanlinXie, 2019-03-19:
Reproduced, @junjiemao1 will follow up.

junjiemao1, 2019-04-03:
@rossburton The board names have been unified so that the configuration files can be properly located. Would you please verify if it works for you?

Mingyuan18, 2019-04-09:
@rossburton , The patch was merged, Would you please verify if it works for you?

Mingyuan18, 2019-04-16:
@rossburton , This PR's tag is : acrn-2019w12.4-140000f; Please help to confirm. Thanks!

fuzhongl, 2019-05-24:
@rossburton   The issue is fixed since acrn-2019w12.4-140000f. Could you please close this ticket?
Thanks!

fuzhongl, 2019-05-28:
@rossburton 
Any more concern, please update the comments. If no more feedback within two weeks, this ticket will be closed by default. 
Appreciate for your understanding.

closed 2019-06-02 by rossburton