projectacrn/acrn-hypervisor #2795: Cannot start UOS

rossburton, 2019-03-15:
I have a X-based SOS and Weston-based UOS that works fine with acrn-hypervisor 0.6.  Upgrading to 0.7, the UOS won't start:

```
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
SW_LOAD: get kernel path /var/lib/machines/core-image-weston.bzImage
SW_LOAD: get bootargs root=/dev/vda rw rootwait maxcpus=1 nohpet console=tty0 console=hvc0   console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M   consoleblank=0 tsc=reliable   i915.nuclear_pageflip=1 i915.avail_planes_per_pipe=0x070F00
VHM api version 1.0
open hugetlbfs file /run/hugepage/acrn/huge_lv1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:0/0 page size:0x200000
level 1 free/need pages:0/2 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 0): echo 2 > /sys/kernel/mm/hugepages/hugepages-1048576kB/nr_hugepages
now enough free pages are reserved!

try to setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x0, highmem 0x0
	level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x0
total_size 0x180000000

mmap ptr 0x0x7fc79e802000 -> baseaddr 0x0x7fc7c0000000
mmap 0x80000000@0x7fc7c0000000
touch 2 pages with pagesz 0x40000000

really setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x0, highmem 0x0
	level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x0
polling 33...
Listening 33...
***********************************************
virt-console backend redirected to /dev/pts/1
***********************************************
tpm: init_vtpm2:Invalid socket path!
/tmp/dm.XznMAIM     30: [0004]		Address : fec00000
Error    6302 -  Flag value is too large ^  (Maximum 1 bit)

/tmp/dm.XznMAIM     37: [0004]		Interrupt : 00000002
Error    6302 -    Flag value is too large ^  (Maximum 1 bit)

/tmp/dm.XznMAIM     46: [0004]		Interrupt : 00000009
Error    6302 -    Flag value is too large ^  (Maximum 2 bit)

/tmp/dm.XznMAIM     47: [0002]		Flags (decoded below) : 000D
Error    6302 -                Flag value is too large ^  (Maximum 2 bit)

vtnet tx thread closing...
mngr_client_new 113
Stop listening 33...
Stop polling 33...
```

Vshmuk, 2019-03-15:
Hi!
I think, i am suffering from the same or a similar bug. Would you please show me your command line parameters?

gvancuts, 2019-03-15:
> Hi!
> I think, i am suffering from the same or a similar bug. Would you please show me your command line parameters?

We have been exchanging on this on the mailing list - see the starting point here: https://lists.projectacrn.org/g/acrn-users/message/262

There was an issue that got fixed in between 0.6 and 0.7 - see https://github.com/projectacrn/acrn-hypervisor/issues/2568

This is not really the same as @rossburton describes as he's running 0.7 which *does* include the fix to that problem.

My suspicion at this stage is that he could have a version of `/usr/sbin/iasl` in his Service OS that is causing the problem. We tested the `accrn-dm` with `iasl` version 20190215. I don't know what version he's running. What about your set-up?

Hacker-vision, 2019-03-16:
Hi! I follow the using_ubuntu_as_sos.rst to build ubuntu 18.04 sos + uos using acrn-hypervisor 0.6.  I am suffering from the same  bug when "sudo /usr/share/acrn/samples/nuc/launch_uos.sh" and uos won't start, even if I have replaced the iasl with the 20190215 from https://github.com/acpica/acpica.

My sos kernel：org.clearlinux.iot-lts2018-sos.4.19.0-22
My uos kernel：clear-26440-kvm.img and linux-iot-lts2018-4.19.0-22.x86_64.rpm.

"Error    6302 -  Flag value is too large ^  (Maximum 1 bit) " is  actually the iasl problem, but I cannot find any solution. Would you please give me any suggestion?

rossburton, 2019-03-16:
We're using acpica 20180508, I can try upgrading to 2190215.

gvancuts, 2019-03-16:
> Hi! I follow the using_ubuntu_as_sos.rst to build ubuntu 18.04 sos + uos using acrn-hypervisor 0.6. I am suffering from the same bug when "sudo /usr/share/acrn/samples/nuc/launch_uos.sh" and uos won't start, even if I have replaced the iasl with the 20190215 from https://github.com/acpica/acpica.

`acrn-dm` will, by default, use `/usr/sbin/iasl` whereas I believe the default installation path when you build it directly is `/usr/bin/iasl`. Can you run `/usr/sbin/iasl -v` and paste the output here?

If different than 2190215, try to put the newer binary under `/usr/sbin` and try again.

rossburton, 2019-03-16:
We fix the iasl path already.  I'l try the new one on Monday.

Hacker-vision, 2019-03-17:
Intel ACPI Component Architecture
ASL+ Optimizing Compiler version 20190215
Copyright (c) 2000 - 2019 Intel Corporation

I'm sure that iasl version is 20190215. So I changed the kernel version with  ACRN v0.7 + SOS 28260 + UOS 26440 from https://lists.projectacrn.org/g/acrn-users/topic/30443828#262. And now I can run uos succssfully with iasl 20180105.

However, I'm still suffering from the same bug using iasl 20190215.

rossburton, 2019-03-18:
Okay, for me it works with acpica 20190215.  Would you consider incorporating a fix so acrn works with both?

Mingyuan18, 2019-03-28:
Hi @rossburton , Would you like to submit a Patch related to this question?

yifanddd, 2019-04-12:
[External_System_ID] ACRN-3372

Mingyuan18, 2019-05-05:
@rossburton , If this problem doesn't exist, can you close it. Thanks!

rossburton, 2019-05-08:
Still a problem.

fuzhongl, 2020-06-28:
@rossburton  acpica version should match with  ACRN-DM version.
It is fine if  acpica is newer than 20190816.

closed 2020-06-28 by fuzhongl