projectacrn/acrn-hypervisor #2798: Cannot stop UOS

Vshmuk, 2019-03-16:
I cannot stop running UOS which is in Preloader mode:

```
# acrn-dm -v
DM version is: 0.7-unstable-”2019w11.4.140000p_172” 
# acrnctl list
vm2		started
testacrn		started
# acrnctl stop vm2
Error happens when try to stop vm. errno(-1)
```

In the VM Console i see the following:
```
====================Linux Loader====================

Press any key within 1 second(s) to enter the command shell
Boot options (in HEX):

Idx|DevType|Flags|DevAddr |HwPart|FsType|SwPart|File/Lbaoffset
  0| VIRTUAL_MEDIA |   1 |     300|    0 |  RAW |    0 | 0x8000
  1| VIRTUAL_MEDIA |   0 |     300|    0 | EXT2 |    0 | iasImage
  2|   MEM |   0 |       0|    0 |  RAW |    0 | 0x3CE0000
  3|   MEM |   2 |       0|    0 |  RAW |    0 | 0x3CE0000

BootMediumPciBase(0x0)
Getting boot image from... MEM
Try to find boot partition
Partition type: UNKNOWN  (1 logical partitions)
Find partion success
Load image from SwPart (0x0), LbaAddr(0x3CE0000)
IAS image is signed!
Connected:35
            Disconnect(35)!
Connected:43
            Connected:35
                        Disconnect(43)!
No handler for id:3
Disconnect(35)!
Connected:44
            Disconnect(44)!
Connected:35
            Disconnect(35)!
Connected:43
            Disconnect(43)!
Connected:35
            No handler for id:3
Disconnect(35)!
Connected:43
            Disconnect(43)!
Connected:35
            Connected:43
                        Disconnect(35)!
No handler for id:3
```

Here is the acrn-dm command line:
```
acrn-dm -A -m 512M -c 1 -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio \
  -s 2,pci-gvt -G "64 448 8" \
  -s 5,virtio-console,@pty:pty_port \
  -s 6,virtio-hyper_dmabuf \
  -s 3,virtio-blk,b,/home/clear/launch_android/data/android/android.img \
  -s 7,virtio-blk,/home/clear/launch_android/hdd.img \
  -s 4,virtio-net,tap0 \
  --mac_seed $mac_seed \
  --vsbl /usr/share/acrn/bios/VSBL_debug.bin \ 
  2
}
```

Vshmuk, 2019-03-18:
Sorry, i may have an older version of the Hypervisor. Will recompile and check if it persists

Mingyuan18, 2019-03-22:
@Vshmuk So can you close this issue? Or you'll have ongoing updates

Vshmuk, 2019-03-22:
I cannot check it due to a new bug in my setup:
https://github.com/projectacrn/acrn-hypervisor/issues/2807

Vshmuk, 2019-03-23:
Okay.
Version:
```
sudo opt/acrn-dm -v
DM version is: 0.8-unstable-19c53425 (daily tag:acrn-2019w11.5-140000p), build by diver@2019-03-18 17:46:26
```

How to reproduce:
```
 sudo opt/acrnctl stop testacrn
Error happens when try to stop vm. errno(-1)
```

VM Startup script:
```
/home/clear/opt/acrn-dm -A -m 512M -c 2 -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio \
  --vsbl /usr/share/acrn/bios/VSBL_debug.bin \
  -s 2,pci-gvt -G "64 448 8" \
  -s 5,virtio-console,@pty:pty_port \
  -s 6,virtio-hyper_dmabuf \
  --enable_trusty \
  -s 3,virtio-blk,b,/home/clear/launch_android/data/android/android.img \
  -p 0:1 \
  testacrn
```

It happens when the VM is stuck in a VSBL bootloader mode:
```
Press any key within 1 second(s) to enter the command shell
Boot options (in HEX):

Idx|DevType|Flags|DevAddr |HwPart|FsType|SwPart|File/Lbaoffset
  0| VIRTUAL_MEDIA |   1 |     300|    0 |  RAW |    0 | 0x8000
  1| VIRTUAL_MEDIA |   0 |     300|    0 | EXT2 |    0 | iasImage
  2|   MEM |   0 |       0|    0 |  RAW |    0 | 0x3CE0000
  3|   MEM |   2 |       0|    0 |  RAW |    0 | 0x3CE0000

BootMediumPciBase(0x0)
Getting boot image from... MEM
Try to find boot partition
Partition type: UNKNOWN  (1 logical partitions)
Find partion success
Load image from SwPart (0x0), LbaAddr(0x3CE0000)
IAS image is signed!
Connected:35
            Disconnect(35)!
Connected:41
            Disconnect(41)!
Connected:35
            Disconnect(35)!
Connected:41
            Connected:35
                        Disconnect(41)!
No handler for id:3
Disconnect(35)!
Connected:42
            Connected:35
                        Disconnect(42)!
No handler for id:3
Disconnect(35)!
Connected:41
            Disconnect(41)!
Connected:35
            No handler for id:3
Disconnect(35)!
```

Mingyuan18, 2019-04-08:
@Vshmuk #2807 has been closed, can this be closed?

yifanddd, 2019-04-12:
[External_System_ID] ACRN-3371

Vshmuk, 2019-04-12:
Hi! No, i still experience this bug. This comment - https://github.com/projectacrn/acrn-hypervisor/issues/2798#issuecomment-475867699 - is actual

fuzhongl, 2019-05-06:
> I cannot stop running UOS which is in Preloader mode:
> 
> ```
> # acrn-dm -v
> DM version is: 0.7-unstable-”2019w11.4.140000p_172” 
> # acrnctl list
> vm2		started
> testacrn		started
> # acrnctl stop vm2
> Error happens when try to stop vm. errno(-1)
> ```
> 
> In the VM Console i see the following:
> 
> ```
> ====================Linux Loader====================
> 
> 
> Press any key within 1 second(s) to enter the command shell
> Boot options (in HEX):
> 
> Idx|DevType|Flags|DevAddr |HwPart|FsType|SwPart|File/Lbaoffset
>   0| VIRTUAL_MEDIA |   1 |     300|    0 |  RAW |    0 | 0x8000
>   1| VIRTUAL_MEDIA |   0 |     300|    0 | EXT2 |    0 | iasImage
>   2|   MEM |   0 |       0|    0 |  RAW |    0 | 0x3CE0000
>   3|   MEM |   2 |       0|    0 |  RAW |    0 | 0x3CE0000
> 
> BootMediumPciBase(0x0)
> Getting boot image from... MEM
> Try to find boot partition
> Partition type: UNKNOWN  (1 logical partitions)
> Find partion success
> Load image from SwPart (0x0), LbaAddr(0x3CE0000)
> IAS image is signed!
> Connected:35
>             Disconnect(35)!
> Connected:43
>             Connected:35
>                         Disconnect(43)!
> No handler for id:3
> Disconnect(35)!
> Connected:44
>             Disconnect(44)!
> Connected:35
>             Disconnect(35)!
> Connected:43
>             Disconnect(43)!
> Connected:35
>             No handler for id:3
> Disconnect(35)!
> Connected:43
>             Disconnect(43)!
> Connected:35
>             Connected:43
>                         Disconnect(35)!
> No handler for id:3
> ```
> 
> Here is the acrn-dm command line:
> 
> ```
> acrn-dm -A -m 512M -c 1 -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio \
>   -s 2,pci-gvt -G "64 448 8" \
>   -s 5,virtio-console,@pty:pty_port \
>   -s 6,virtio-hyper_dmabuf \
>   -s 3,virtio-blk,b,/home/clear/launch_android/data/android/android.img \
>   -s 7,virtio-blk,/home/clear/launch_android/hdd.img \
>   -s 4,virtio-net,tap0 \
>   --mac_seed $mac_seed \
>   --vsbl /usr/share/acrn/bios/VSBL_debug.bin \ 
>   vm2
> }
> ```
Could you please share more information about preloader mode?

fuzhongl, 2019-05-08:
What the information about acrnctl list? the VM is in running or not?
Stop command is used to stop one or more running VM. 
About the OS boot process is not successful for any reason, could you please try acrnctl del command first? then acrnctl add
Thanks!

Vshmuk, 2019-05-09:
> What's the information about acrnctl list? the VM is in running or not?
As i mentioned here: https://github.com/projectacrn/acrn-hypervisor/issues/2798#issuecomment-489543301
Yes, it's running.

> Stop command is used to stop one or more running VM.
> About the OS boot process is not successful for any reason, could you please try acrnctl del command first? then acrnctl add.
I cannot del before i stop. But 'stop' command seems to be not working when VM is in the bootloader
> Thanks!

fuzhongl, 2019-05-09:
We try to reproduce the issue:
What's the platform and fail rate?
Could you please paste the launch.sh(if it is modified on your side)? 
Thanks!

Vshmuk, 2019-05-10:
Hi!
> What's the platform and fail rate?
It was  NUC model nuc7i7bnh. Fail rate - always.
> Could you please paste the launch.sh(if it is modified on your side)?
Already no, since i don't have access to the device anymore. Anyway, I pasted the command line for running acrn-dm. But except this command, there was only a loop to turn off additional CPUs.

fuzhongl, 2019-05-10:
> /home/clear/opt/acrn-dm -A -m 512M -c 2 -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio \
>   --vsbl /usr/share/acrn/bios/VSBL_debug.bin \
>   -s 2,pci-gvt -G "64 448 8" \
>   -s 5,virtio-console,@pty:pty_port [...]

For NUC platform, there is no ,b option for virtio-blk.
Please check: https://github.com/projectacrn/acrn-hypervisor/issues/1917  
Thanks

Vshmuk, 2019-05-11:
Okay. So?

fuzhongl, 2019-05-13:
> Okay. So?

Not sure where your acrn-dm coming from?
Since the UOS isn't launched successfully, please fix your acrn-dm option first. You can refer to the ''launch_uos.sh''(under acrn-hypervisor\devicemodel\samples\nuc)
Thanks!

gvancuts, 2019-05-13:
> For NUC platform, there is no ,b option for virtio-blk.
> Please check: #1917

The information regarding the `,b` option mentions it's to be used with `vSBL`, which is the case here.

binbinwu1, 2019-05-13:
Is the platform KBL based?

gvancuts, 2019-05-13:
> Is the platform KBL based?

Yes, see https://github.com/projectacrn/acrn-hypervisor/issues/2798#issuecomment-491295914

binbinwu1, 2019-05-13:
vSBL currently doesn't support KBL.

Vshmuk, 2019-05-13:
Okay, i agree that my command line syntax may be wrong since the platform does not support some stuff. But the main problem is not with syntax or stuff support or whatever. The problem is that _i cannot turn off the virtual machine_. For me, it must always work regardless of the correctness of syntax to run

fuzhongl, 2019-05-14:
> Okay, i agree that my command line syntax may be wrong since the platform does not support some stuff. But the main problem is not with syntax or stuff support or whatever. The problem is that _i [...]
For "acrnctl stop" my understanding is: UOS should be launched successfully first.
Are we on the same page? If yes, please close this issue.
Thanks!

gvancuts, 2019-05-14:
> For "acrnctl stop" my understanding is: UOS should be launched successfully first.

So what is the recommended method for shutting down a VM (that has had problems booting up but nevertheless listed as `started`).

fuzhongl, 2019-05-14:
> > For "acrnctl stop" my understanding is: UOS should be launched successfully first.
> 
> So what is the recommended method for shutting down a VM (that has had problems booting up but nevertheless listed as `started`).

For this case, my choice is reboot SOS(ssh to SOS). 
Thanks!

gvancuts, 2019-05-14:
> For this case, my choice is reboot SOS(ssh to SOS).

This does not sound like a good, robust solution. We should treat that as a bug and make `acrnctl` more resilient to situations like these.

fuzhongl, 2019-05-14:
Ok, will forward this issue to the component owner.

fuzhongl, 2019-05-14:
@Vshmuk Following is the direction from component owner, please!
Could you please capture log to a file and upload as attachment? It's hard to check the log paste like this.

gvancuts, 2019-05-14:
> @Vshmuk Following is the direction from component owner, please!
> Could you please capture log to a file and upload as attachment? It's hard to check the log paste like this.

Not sure what logs you want. @Vshmuk has indicated he does not have his setup anymore. But the good news is that it's very easy to reproduce. Just follow these few steps:
* Install latest Clear Linux and ACRN (CL 29400, acrn 1.1-unstable) on a KBL NUC
* Follow all the steps to run a LaaG (and make sure it can be run correctly)
* Modify the `launch_uos.sh` script by adding `--vsbl VSBL.bin` to it

Run that new script and see that it fails to boot the VM. At that point, `acrnctl list` will report that that VM is `started` but it's impossible to stop, reset or delete it. Note that I was also unable to launch any new VM after that, so a complete reset of the platform was necessary.

Vshmuk, 2019-05-15:
@gvancuts Thanks for the support!
I am not aware of how the hypervisor works, but for me, it's usual to have two shutdown options. The first one is ACPI call to the OS, which is soft, but may not work in case the guest OS is broken (my case), and the second one is 'destroy', which shuts it off completely regardless what is inside of the container.

fuzhongl, 2019-05-29:
@Vshmuk 
Following command can be used to stop UOS for your case:
kill -2 <acrn_dm_pid>
Make sense?
If yes, please help to close this ticket.
Appreciated for your understanding.

gvancuts, 2019-08-27:
@fuzhongl a new `-f/--force` option was recently added to the `acrnctl stop` command. Do you test it using this scenario?

closed 2019-08-27 by fuzhongl

fuzhongl, 2019-08-27:
@gvancuts Not try --force option for acrnctl stop; will check with developer for detail.
Thanks!