projectacrn/acrn-hypervisor #2799: Cannot start UOS with Android  kernel

Vshmuk, 2019-03-16:
Version:
```
# acrn-dm -v
DM version is: 0.7-unstable-”2019w11.4.140000p_172” 
```

Config:
```
/usr/bin/acrn-dm -A -m 512M -c 2 -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio \
  -s 2,pci-gvt -G "64 448 8" \
  -s 5,virtio-console,@pty:pty_port \
  -s 6,virtio-hyper_dmabuf \
  --enable_trusty \
  -p 0:1 \
  -k /home/clear/android_kvm/fast_flashfiles/boot.img-kernel.gz \
  testacrn
```

I see a strange log in VM stdout:
```
VHM api version 1.0
open hugetlbfs file /run/hugepage/acrn/huge_lv1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:256/256 page size:0x200000
level 1 free/need pages:0/0 page size:0x40000000

try to setup hugepage with:
	level 0 - lowmem 0x20000000, biosmem 0x0, highmem 0x0
	level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
total_size 0x140200000

mmap ptr 0x0x7f5670fac000 -> baseaddr 0x0x7f5671000000
mmap 0x20000000@0x7f5671000000
touch 256 pages with pagesz 0x200000

really setup hugepage with:
	level 0 - lowmem 0x20000000, biosmem 0x0, highmem 0x0
	level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
polling 33...
             Listening 33...
                            ***********************************************
                                                                           virt-console backend redirected to /dev/pts/4
                                                                                                                        ***********************************************
                   virtio_hyper_dmabuf: Ooops! Re-entered!!
                                                           virtio_hyper_dmabuf: VBS-K init failed with error -1!
                                                                                                                tpm: init_vtpm2:Invalid socket path!
                                                                                                                                                   SW_LOAD: kernel /home/clear/android_kvm/fast_flashfiles/boot.img-kernel.gz size 10895360 copied to guest 0x1000000
                                                                                                                 SW_LOAD: found setup sig @ 0x000042c8, setup_size is 0x00004400
                            SW_LOAD: build zeropage for bootargs addr: 0x1fffe000
                                                                                 SW_LOAD: build e820 7 entries to addr: 0x7f5690fff2d0
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000ef000,  type 0x1
SW_LOAD: entry[1]: addr 0x00000000000ef000, size 0x0000000000011000,  type 0x2
SW_LOAD: entry[2]: addr 0x0000000000100000, size 0x000000001ff00000,  type 0x1
SW_LOAD: entry[3]: addr 0x0000000020000000, size 0x0000000060000000,  type 0x2
SW_LOAD: entry[4]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[5]: addr 0x0000000100000000, size 0x0000000040000000,  type 0x2
SW_LOAD: entry[6]: addr 0x0000000140000000, size 0x0000000000100000,  type 0x2
SW_LOAD: zeropage prepared @ 0x1ffff000, kernel_entry_addr=0x1004400
                                                                    full reset
VM loop exit
            mngr_client_new: Failed to accept from fd 33, err: Invalid argument
Stop listening 33...
Stop polling 33...
```

coneypo, 2019-03-18:
Please add these informations so that we can reproduce this issue in our environment and try to solve this issue, thanks！

**HW/Board**
APL_NUC or/and KBL_NUC or/and other platforms

**Build link**

0.7-unstable-”2019w11.4.140000p_172” 

**Image info**

28320 (Clear Linux Version)

**Steps**

swupd update -m 28320 .

**Expected result**

Start UOS with Android kernel

**Actual result**

Cannot start UOS with Android kernel

**Reproduce rate**
x/x

Vshmuk, 2019-03-18:
Sorry, i may have an older version of the Hypervisor. Will recompile and check if it persists

Vshmuk, 2019-03-18:
https://github.com/projectacrn/acrn-hypervisor/issues/2807

closed 2019-03-18 by Vshmuk